freifunk-gluon/gluon #2175: Multiple vulnerabilities in dnsmasq < 2.83

goligo, 2021-01-20:
Multiple security issues have been reported for dnsmasq, see

https://www.heise.de/news/DNSpooQ-Sicherheitsluecken-im-DNS-DHCP-Server-Dnsmasq-entdeckt-5030266.html

Issues are fixed with version 2.83. Used version in 2020.2.2 is 2.80.

As far I understand, dnsmasq is only used locally within the node, and is neither exposed to the Freifunk network, nor to the local network, so there should be no risk and so no urgency to update it.

mweinelt, 2021-01-20:
Yes, we are aware of these issues. They affect DNSSEC for the RCE and Caching for the Cache Poisoning, both of which are not enabled on Gluon nodes, so we don't expect any impact.

blocktrron, 2021-01-20:
As we are only affected by CVE-2020-25684, i don't think it is that much of a threat for us. 

master as well as v2020.2.x already contain the necessary version updates from upstream. They will be included in the next Gluon release.

goligo, 2021-01-20:
Thanks, I guess I can close the issue, as no further action is required.

closed 2021-01-20 by goligo