freifunk-gluon/gluon #2176: Ubiquiti ER-X SFP shows (private) WLAN settings and VPN key N/A

2tata, 2021-01-23:
The ER-x SFP shows wifi settings. Which should not shown as the device does not have any Wifi chip.

Another minor issue the VPN key on the information screen and on the reboot page is always shown as N/A.

[image]

[image]

[image]

blocktrron, 2021-01-23:
@2tata regarding the VPN key: Which VPN provider (tunneldigger / fastd) are you using?

2tata, 2021-01-23:
@blocktrron I have updated the above discription

AiyionPrime, 2022-01-25:
The vpn key does not lead to problems with wireguard;
but I can confirm the wifi-related-issue on an erx (no sfp)

gluon-v2021.1-148-gb7c343ac
gluon: 3f7c0b3ae8297541389a2512ac92ad5c1a02bbeb
site: https://github.com/freifunkh/site/commit/cb027748417b4e88fdbe96919feb213e31e4242c

NeoRaider, 2022-01-25:
The VPN key works as expected: Tunneldigger doesn't have a key, so "n/a" is displayed.

The WLAN part was fixed with #2312 (#2311 was actually a duplicate of this issue).

closed 2022-01-25 by NeoRaider

AiyionPrime, 2022-01-25:
Was about to revoke this, thx. Saw #2312 just now. Sorry for the noise.