freifunk-gluon/gluon #2180: Bug: Gluon Routes its Packets to Wrong IPv6 Default Routers

lemoer, 2021-02-19:
### Bug report

**What is the problem?**

Router advertisement servers advertise *prefixes* together with a *default router* in layer 2 networks. There may be multiple router advertisement servers in one layer 2 network. Each of those router advertisement servers can announce different *prefixes* and a different *default router*.

A client can accept multiple of the advertised *prefixes* at the same time. If it does that, it installs an address in each of the advertised prefixes. It also has to install default routes from the router advertisements. Since default routers are not expected to be able to forward ip packets from arbitrary source addresses, the client should install routes to all *default routers* and use the correct one depending on which of the installed addresses it is sending from.

A gluon node receives all router advertisements on the `br-client`. It accepts all *prefixes* and therefore installs one address in each received *prefix*. But it installs just one *default route*. Therefore it sends all packets to the same *default router*. As this *default router* is not the designated *default router* for all prefixes, it may happen that the packet is dropped because of its source ip. This means that the gluon node will not be publicly reachable on some of its addresses.

**What is the expected behaviour?**

Gluon nodes should either:
- accept only one router advertisement at a time,
- or should make sure, that when they are sending packets, they use the correct default route based on the source address.

**How do you recognize, that a gluon node is affected?** 
 
- The gluon node will not be publicly reachable on some of its addresses.

**What are the conditions, this bug is triggered:**
- Multiple router advertisement servers are in the mesh network.
- These servers advertise different *default routers* and different *prefixes*.
- The *default routers* are not able to route packets with source ips of other *prefixes*.

**Not affected:**
- Clients of the gluon node are not affected. The *radv-filterd* is filtering here.
- Forwarding is not affected. This is only about the packets, whose source is the gluon node itself. 
- IPv4.

**Possible workarounds:**
- Implement source based routing between the *default routers*:
    - Linux has supported this without policy-based routing for quite some time, but documentation on the Internet is very sparse. (It's almost not findable via google.)
    - Using the ip-tools: `ip route add ::/0 from PREFIX via OTHER_ROUTER dev IFACE`.
    - In systemd-networkd, the `Source=...` can be used inside the `[Route]` section.

**Gluon Version:**
- master

**Site Configuration:**
- n/a

**Custom patches:**
- n/a

lemoer, 2021-02-19:
We probably found the broken code:

https://github.com/openwrt/openwrt/blob/master/package/network/ipv6/odhcp6c/files/dhcpv6.script#L134

Actually the code tries to be quite clever here, if I interpret it correctly. It applies source source routing only if necessary. Only if there "duplicate" routes, then it applies source based routing if not, it does not apply source based routing. But there is a mistake in L134. It should be `"$duplicate" = 0` instead of `"$duplicate" = 1`.

Before any changes
```
root@wgtest-1043-lemoer:/lib/netifd# ip -6 r | grep default
default via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
```

After changing it to  `"$duplicate" = 0`, we get:
```
root@wgtest-1043-lemoer:/lib/netifd# ip -6 r | grep default
default from 2001:678:978:114::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
default from 2a02:790:ff:714::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
default from 2a02:790:ff:814::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
default from 2a02:790:ff:914::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
default from 2a02:790:ff:1014::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
default from fdca:ffee:8:14::/64 via fe80::8ae6:40ff:feba:1014 dev br-client proto static metric 512 pref medium
```

awlx, 2021-02-19:
According to RFC3484 there are only 8 rules to select the source address, other than that you need extra logic. Or just route between your gateway. So I wouldn't call this a bug but default RFC behaviour. It might not be desired and expected but it's fine :/.

awlx, 2021-02-19:
There was a [draft](https://www.ietf.org/archive/id/draft-pfister-6man-sadr-ra-01.txt) for IPv6 to support what you want. But according to the actual current [RFC6724](https://tools.ietf.org/html/rfc6724) this is not part of IPv6 Default Address selection. Thus it's not a bug but maybe a missing feature. All gateways in a Layer2 segment should be able to route all addresses.

lemoer, 2021-02-20:
Thanks for the clarification and the references. From the RFC point of view, this is maybe not a bug.

If I understand the odhcpv6 script correctly, actually wants to implement what we need. But unfortunately, there is a bug in an if statement. When I invert `"$duplicate" = 1` to `"$duplicate" = 0`, the problems are solved. I updated my comment above to (hopefully) make this more clear.

awlx, 2021-02-20:
Btw just to mention, even this does not help you getting rid of all cases where wrong prefixes hit the wrong gateways. Imagine a roaming client with the „wrong prefix“ which is still valid until valid lifetime expires. And it receives another RA with the other Gateway when connected to another node. Then you have the same problem. Thus you have to route between the gateways nevertheless.

CodeFetch, 2021-02-20:
@awlx Yes, that there is still routing between the supernodes is clear I think, but I'm also sure that this is a bug. As far as I know support for multiple IPv6 prefixes using source routing is needed since a revision of the IPv6 protocol done for the Homenet protocol and it is part of the mainline kernel since 4.1.

Adorfer, 2021-02-20:
Even in case we consider this behaviour
1) not to be a bug (of the ipv6 implementation in openwrt), and
2) as defined / in compliance with RFCs

It should be changed (at least in batman-adv gluon) in order to reduce "diagonal traffic" in multi-supernode L2 segments. 
Since this this type of traffic reduces overall performance of the system. (or increases performance, if it can be avoided)

awlx, 2021-02-20:
> @awlx Yes, that there is still routing between the supernodes is clear I think, but I'm also sure that this is a bug. As far as I know support for multiple IPv6 prefixes using source routing is [...]

I have never seen this working yet with pure Slaac on Linux also Homenet mentions the same RFCs I mentioned. https://tools.ietf.org/id/draft-ietf-homenet-arch-04.html#rfc.section.2.1

We also run multiple prefixes at FFMUC and I can confirm that neither MacOS or most Linux Distros do any kind of source routing they just send their stuff where they want.

lemoer, 2021-02-20:
This is acutally not a bug in OpenWrt. We are just disabling the feature in gluon. I created a pull request in #2183 to enable source routing.

CodeFetch, 2021-02-21:
From RFC 6724 - Default Address Selection for IPv6 

```
Rule 5.5: Prefer addresses in a prefix advertised by the next-hop.
   If SA or SA's prefix is assigned by the selected next-hop that will
   be used to send to D and SB or SB's prefix is assigned by a different
   next-hop, then prefer SA.  Similarly, if SB or SB's prefix is
   assigned by the next-hop that will be used to send to D and SA or
   SA's prefix is assigned by a different next-hop, then prefer SB.

      Discussion: An IPv6 implementation is not required to remember
      which next-hops advertised which prefixes.  The conceptual models
      of IPv6 hosts in Section 5 of [RFC4861] and Section 3 of [RFC4191]
      have no such requirement.  Hence, Rule 5.5 is only applicable to
      implementations that track this information.
```

This sounds to me like SHOULD, but not MUST. So maybe we could tune the metrics...

CodeFetch, 2021-02-21:
I think we could introduce a "RAforeignprefixmetricpenalty" option which adds a penalty to the metric of a route with the "from" of a prefix which was not advertised by that specific gateway. I think this should be conform to the homenet specification, too.

NeoRaider, 2021-02-21:
As mentioned in #2183, I currently don't see how we can support IPv6 routers announcing different prefixes.

Most clients don't implement Source Routing, so if radv-filterd only allows RAs from one Router to pass through, and a client roams to a different node where radv-filterd has chosen a router with a different prefix, things will break badly. As such, all routers that announce a default route at all must be equivalent, in that they can route packets from/to the same address space. And as long as this is the case, it doesn't matter which default route Gluon uses.

For networks that don't have their own IP space, and can't or don't want to have BGP peering on each router, a common solution that was also used by FFHL in the past is Network Prefix Translation (NPTv6). This is not perfect, as connections will still break when a client roams to a different default route, but at least no client retains IP addresses that aren't usable anymore for an indefinite amount of time.

CodeFetch, 2021-02-21:
@NeoRaider 

`all IPv6 routers must ~announce~ be able to route the same set of prefixes`

I'm aware of that. It means that we must add a route for every gateway nonetheless, but they should have a worse metric if that "from" prefix was not announced by that specific gateway, because it means the packet must be re-routed in the backbone and therefore it is logical that the metric should have added a penalty for which we do not account, yet.

My suggestion is to add an option to the dhcp6c script for defining a metric penalty which will be applied for source routes which use a prefix which was not announced by that specific gateway.

My thought: The only reason why we don't open up a VPN connection to every supernode is that we get the broadcast and Batman management traffic for each. If Batman or when using Babel mmfd would use a more sophisticated multicast approach, we could limit the amount of duplicate management traffic which allows us to get rid of most cross-traffic on the supernodes and allows us to do more decentralization in the future.

CodeFetch, 2021-02-21:
> Most clients don't implement Source Routing, so if radv-filterd only allows RAs from one Router to pass through, and a client roams to a different node where radv-filterd has chosen a router with a different prefix, things will break badly.

I'm not suggesting to stop routing between the supernodes...

NeoRaider, 2021-02-21:
> @NeoRaider
> My suggestion is to add an option to the dhcp6c script for defining a metric penalty which will be applied for source routes which use a prefix which was not announced by that specific gateway.

I think the `dhcpv6.script` or netifd will need to choose one default route as the "primary" one and install that route without a source restriction, so it will be used for source address selection. Having this in netifd might make sense for route timeout?

awlx, 2021-02-24:
To be honest, I wouldn't fiddle with default Linux behaviour as none standard behaviour is hard to debug. We basically provoke this problem ourselves with stretching L2 domains over different datacenters.

CodeFetch, 2021-02-24:
@awlx We are the people who define Linux default behavior. If we don't, never will anyone else, especially when it comes to routers.

awlx, 2021-02-24:
@CodeFetch Then I would open a bug way upstream in the kernel and not here. Otherwise it's just a "variant of Linux" which behaves completely different.

CodeFetch, 2021-02-24:
@awlx I think the kernel already has all features implemented. 
In [this line](https://github.com/openwrt/odhcp6c/blob/53f07e90b7f1da6977143a488dd5cb73a33b233b/src/ra.c#L486) the router is set to "any" and the information is not propagated to the [script here](https://github.com/openwrt/openwrt/blob/dc8cc446128776ee69571ff03925e6bd3256c2e0/package/network/ipv6/odhcp6c/files/dhcpv6.script#L136).

This is correct according to the standard, but if possible the prefix that was actually announced by the gateway should be preferred (by setting e.g. a metric penalty for the routes with the prefixes where the gateway didn't advertise the prefix itself - as we can assume that it needs to be rerouted in the backbone this is factually correct). I just propose to introduce a configurable metric penalty for these routes. This should not break anything and is still conform to the IPv6 standard. 

```
      Hence, Rule 5.5 is only applicable to
      implementations that track this information.
```

I think we should implement it.