freifunk-gluon/gluon #2182: gluon-core: entropy daemon not available on all platforms

blocktrron, 2021-02-20:
### Bug report

**What is the problem?**

urngd is not compatible with every platform. The timer is to coarse to generate jitter entropy from it on some platforms.

The kernel ringbuffer will in this case read

```
[   10.250566] urngd: jent-rng init failed, err: 2
```

This problem exists at least on ramips-mt7620 and bcm-2708.

**What is the expected behaviour?**

Have an entropy daemon for generating random ssh host keys as well as fastd keys.

**Gluon Version:**

Gluon 2020.2

**Site Configuration:**

Not site-dependent if haveged is not included in the site.

**Custom patches:**

-

NeoRaider, 2021-03-07:
Any ideas how to solve this? Does haveged work on these platforms?

blocktrron, 2021-03-15:
Yes, haveged works. AFAIR was haveged only a recommendation in the docs (cit needed), mentioning this in the docs might be fine.

Otherwise, adding a dependency for haveged and removing urngd from the package set is also a viable solution.

NeoRaider, 2021-03-15:
It would be interesting to know if haveged actually produces good entropy on these platforms, and if it does, if urngd can be made to do the same. I can have a look at the code this week, but it's unlikely that I'll find out anything without the hardware in question...

NeoRaider, 2021-03-16:
Upstream `jitterentropy-rngd` used by `urng` recently gained a fallback mode for systems without high-resolution timers. We should evaluate if `urng` can be updated to use the new version without increasing its size too much.

neocturne, 2025-03-29:
Hmm, I don't think I own any mt7620 devices. It would be great if someone could install a current Gluon (2023.2.x should be sufficient) and check if urngd uses a lot of CPU after boot.

Djfe, 2025-04-09:
A fix was just pushed upstream. You might want to compare performance in openwrt main with the fix
https://github.com/openwrt/openwrt/commit/eec11fbbb6eb70cfcae27ac6c055d2c1bd5d20d9

I like the idea of dropping RSA host keys, but fear the implications of locking people out of their devices. Especially devices like the NWA55AXE without a reset button. (yes, you can still fix this serially, but that not so nice with a device attached to some mast outdoors)

brainstorming workarounds (probably way to specific):
We could add a script that increases the major version by one in-case no RSA key is installed on the device to allow updating it to a new major version release without RSA.
Alternatively the script could increase major version if atleast one ED25519 key is present.
Alternatively this could only be done/enforced for devices without a reset key. But I dislike that approach

Note: The new sysupgrade doesn't have to contain an image that identifies as a new major release. it's enough if the metadata of the sysupgrade says so. (avoids having to keep this change around forever but would require an intermediate release)

Overall new communities should be ED25519 only. We could add specific settings to site.conf or specific package names to image-customization.lua to allow for communities themselves to switch to this mode while we figure out how to do a transition.

Aachen likely has tons of devices out there with just rsa keys (even if most of them are by Freifunk members we have no way of reaching everyone affected)

I just had an idea for an actual alternative to the above (increase of the major version):
Add entries to site.json for alternative update paths for devices with no keys or just ED25519-keys.
Those can be used by communities to transition devices into a new release without RSA.
Devices won't accept these no-rsa images over the regular old update path. either due to new metadata or a specific file-name for no-rsa images.

Djfe, 2025-04-09:
this following text was due to a great misunderstandment (host key vs client key), but I'll leave it here in a separate comment but mark it so it stays hidden

but fear the implications of locking people out of their devices. Especially devices like the NWA55AXE without a reset button. (yes, you can still fix this serially, but that not so nice with a device attached to some mast outdoors)

brainstorming workarounds (probably way to specific):
We could add a script that increases the major version by one in-case no RSA key is installed on the device to allow updating it to a new major version release without RSA.
Alternatively the script could increase major version if atleast one ED25519 key is present.
Alternatively this could only be done/enforced for devices without a reset key. But I dislike that approach

Note: The new sysupgrade doesn't have to contain an image that identifies as a new major release. it's enough if the metadata of the sysupgrade says so. (avoids having to keep this change around forever but would require an intermediate release)

Overall new communities should be ED25519 only. We could add specific settings to site.conf or specific package names to image-customization.lua to allow for communities themselves to switch to this mode while we figure out how to do a transition.

Aachen likely has tons of devices out there with just rsa keys (even if most of them are by Freifunk members we have no way of reaching everyone affected)

I just had an idea for an actual alternative to the above (increase of the major version):
Add entries to site.json for alternative update paths for devices with no keys or just ED25519-keys.
Those can be used by communities to transition devices into a new release without RSA.
Devices won't accept these no-rsa images over the regular old update path. either due to new metadata or a specific file-name for no-rsa images.

rotanid, 2025-04-09:
> We may consider disabling RSA host keys for dropbear, as all SSH clients should support Ed25519 these days (support was added to OpenSSH in version 6.5 from 2014).

if that is only for the router's host keys and one can still use RSA as ssh client key?
and would this change generate new keys on all nodes or leave old rsa host keys?

neocturne, 2025-04-09:
@rotanid Yes, host key and client key types are independent.

Currently all nodes have both RSA and Ed25519 keys; I don't see a reason to delete existing RSA keys, so only new/firstbooted nodes would become Ed25519 only. There is no configuration option specifically for key generation at the moment however, so this would require a change in OpenWrt.

herbetom, 2025-04-10:
That would only mean no RSA Host Keys but wouldn't affect the possibilty of using RSA Keys for user authentication?

neocturne, 2025-04-10:
@herbetom Correct.

herbetom, 2025-04-10:
Great. Then i don't see any reason not to do it. :)