freifunk-gluon/gluon #2184: gluon-neighbour-info returns late

AiyionPrime, 2021-03-08:
### Bug report

**What is the problem?**
The tool `gluon-neighbour-info` does return late (until timeout), given the right parameters.

I think it's always been this way, and has just not bothered anyone yet.
It does behave similiary on all my devices spread across several gluon versions. 

**What is the expected behaviour?**
`gluon-neighbour-info -d ::1 -p 1001 -r nodeinfo` should return as fast as the explicit `gluon-neighbour-info -c 1 -d ::1 -p 1001 -r nodeinfo`.

e.g.
`gluon-neighbour-info -d ::1 -p 1001 -r nodeinfo` is expected to return one value, as there is no `-l` parmeter given.
It does print the expected value rather fast, but does not return until the default timeout of `-t`=`3` passed.

Wouldn't it be sensible to let -c default to `1` in case `-l` is not given?

Or are there scenarios, where two structs are returned at once?
I expected `-r neighbours`, but even that does just return one dictionary.

**Gluon Version:**

e.g. v2020.2-187-g57b40f5d
and v2020.1.3-3-g83b67555
but very likely any.

**Site Configuration:**

https://github.com/freifunkh/site/commit/7b398092ab2e504fca9ba6f62cb0063e2545af03
and
https://github.com/freifunkh/site/commit/7b398092ab2e504fca9ba6f62cb0063e2545af03
though it's likely not related to a specific site.

**Custom patches:**

I have not patched anything yet, but testing for the existence of `-l` and letting `-c` default appropriately may be a good start.

NeoRaider, 2021-03-08:
`gluon-neighbour-info` is usually used to get information from neighbours via multicast - the `-d ::1` usecase is somewhat special.

I think we could choose the default behaviour based on the destination address, and exit on the first reply when it is a unicast address.

AiyionPrime, 2021-03-12:
Have you got an idea, how to tackle the is_unicast() part?
Checking if something is multicast is easy, but unicast?

If all fails, it'd be worth something, to check at least for `in6addr_loopback`, so the default `-d` is matched.

NeoRaider, 2021-03-12:
Any valid destination address is either multicast or unicast, so you can just use `!is_multicast()`.

AiyionPrime, 2021-03-12:
`is_multicast()` is implemented and in use.

PR #2185 (2021-03-15): gluon-neighbour-info: end before timeout
This is has been compiled and tested.

Results of `time ./gluon-neighbour-info -s neighbour -d ff02::2:1001 -p 1001 -r nodeinfo`
before and after:

`real	0m3,013s`
and
`real	0m0,168s`

> End the process after one result in case  is not given.
> Reduces singleshot execution time from timeout seconds to around 150ms.
> 
> resolves #2184
commit 2c0f8292 (2021-03-15): gluon-neighbour-info: end before timeout (#2185)

End the process after one result in case -l is not given
and destination address is unicast.
Reduces singleshot execution time from timeout seconds to around 150ms.

resolves #2184