freifunk-gluon/gluon #2193: status-page TQ cut off

AiyionPrime, 2021-04-06:
### Bug report

**What is the problem?**

On gluon master, as well as recent gluon releases the neighbors transmission quality is displayed on the right side of the frame.
When the value reaches `100 %`, the text gets cropped to `100...`.

The problem was introduced sometime after gluon v2020.1.3 (792a93b4dd8841daf6e8ff57be055b12384b2ce2).

The problem occurs on several devices with latest firefox or chromium.

The problem is target independent.

**What is the expected behaviour?**

The cropping appears to be unintended and does not improve readability.
In case this is a problem only for some clients, I can provide screenshots.

**Gluon Version:**

Looking at the commit history of the file commit 016b323a1710cb205ce4862e9f4a8f9cb4db6320 would be my first guess to have introduced this behavior,
containing layout changes of @nalxnet, which would be in v2020.2.2 (90d0e33c619cef9e0af928ef4d6477f6c1bdc0de)

**Site Configuration:**
The problem appears to be site independent, as I found it in testing-releases in ffh 
http://[2001:678:978:116:a2f3:c1ff:fe5e:c8e6]/cgi-bin/status
as well as in ffrn.
http://[2a01:4f8:171:fcff:1ad6:c7ff:fe90:dec5]/cgi-bin/status

nalxnet, 2021-04-07:
Can you attach some screenshots showing this? I can only reproduce this with your example nodes when the window is too narrow to even display more than one character of the neighbour node name, just before the view switches from tablet to mobile (between 746 and 700px).

AiyionPrime, 2021-04-07:
Sure, the image lies on my server in an uncropped version; not sure what github makes of it:
See the issue in the bottom right.

[image]

http://aiyionpri.me/shared/2021-04-07_1920x1080_scrot.png

As the resolution is at about 2K, maybe its just an issue on wide displays.
In firefox there's the `responsive designmode`, which has a profile for "4K Ultra HD Television" which should allow you to reproduce it on smaller screens.

nalxnet, 2021-04-07:
As the table row sizes do not change above 1251px width, this rather seems to be an issue with the font your Firefox is using. It looks slightly wider than the one used on my system. The stylesheet lists Roboto, Lucida Grande, and Arial as the preferred fonts, do you have any of those installed?

If you want to fiddle with it, you can use Firefox' developer tools to increase the width property of the "table.datatable th.row-tq" selector until you see the full value. We currently use 42px, text-overflow starts on my machine (Robot font) for anything below 41px. But I don't see any reason why we shouldn't increase this a little to give the row enough space for people with slightly wider fonts, given they are common on other OS.

AiyionPrime, 2021-04-07:
Firefox inspector shows Arial as used in the fonts tab.
I use Linux though; the package providing is `ttf-liberation` in aur or `liberation-fonts` with apt.

They aim for metric congruence though. I have no machine to verify this happens with regular Arial as well.

belzebub40k, 2021-04-07:
I checked it on my Linux system with Firefox and when I switch the font to Arial it is not cut off.

AiyionPrime, 2021-04-07:
When I install roboto, the issue is gone.

AiyionPrime, 2021-04-07:
> I checked it on my Linux system with Firefox and when I switch the font to Arial it is not cut off.

Arial is currently not chosen by my system; reinstalling would be annoying, due to microsofts EULA...
I'll stick with roboto and hope this was just a bug on my side and that everyone else I saw just lacks these three fonts as well.

@bschelm maybe you might test whether youve got any of them installed.

Is there a way to get rid of this issue independent of the used font?
I think it's only an overflow issue since this was rewritten.

AiyionPrime, 2021-04-07:
The longer I look at it, it's rather annoying how much space on the page is unused; that overflow issue is even more ridiculous in that light...

nalxnet, 2021-04-07:
> Is there a way to get rid of this issue independent of the used font?
> I think it's only an overflow issue since this was rewritten.

As I wrote above, I have no objections to make the TQ column a little bit wider to prevent the text-overflow that seems to happen with other (unselected) fonts. But since the extra space is taken from the node name column, it should only be as much as necessary.

Since you can reproduce it the easiest way would be to increase the value in your Firefox developer tools (called "inspector" or similar) until the overflow is gone and insert it here https://github.com/freifunk-gluon/gluon/blob/2e6752074f159f0a24eaeb98fa2efabc2df17c5d/package/gluon-status-page/sass/status-page.scss#L164. After that, re-generate https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-status-page/files/lib/gluon/status-page/www/static/status-page.css from the edited scss file via your preferred sass compiler and open a PR.

nalxnet, 2021-04-07:
> The longer I look at it, it's rather annoying how much space on the page is unused; that overflow issue is even more ridiculous in that light...

Well, this also has to work for narrow screens. Therefore, the idea is to use as less space as necessary for fixed length values (TQ) and use the rest of the available space for something that can potentially grow very large (node name). See also the original issue where my change originated from: https://github.com/freifunk-gluon/gluon/issues/1960

AiyionPrime, 2021-04-07:
45px appears to work for `DejaVu Sans` which my Firefox seems to default to.

AiyionPrime, 2021-04-07:
I just installed sass as their webpage suggests none of the parameters work;
Is it possible we still use the deprecated version `ruby sass`?

AiyionPrime, 2021-04-07:
Not that it has too much to do with this, but we should get rid of its usage, don't we?
https://sass-lang.com/blog/ruby-sass-is-deprecated

nalxnet, 2021-04-07:
I used sassc last year to compile the scss file IIRC...

AiyionPrime, 2021-04-07:
I tried that first, I still have got the error message in scroll buffer, I think.
`sassc --sourcemap=none -C -t compressed sass/status-page.scss files/lib/gluon/status-page/www/static/status-page.css`
> Invalid argument for -m flag: 'none'. Allowed arguments are: auto inline
See 'sassc -h'

AiyionPrime, 2021-04-07:
I'll report back once I compiled a firmware with it and verified it works as intended for both me and @bschelm.

PR #2196 (2021-04-07): statuspage: fix tq width
...overflow for 'DejaVu Sans'.

This resolves #2193.

This needs testing - I don't usually do web development.
closed 2021-04-07 by mweinelt