freifunk-gluon/gluon #2194: v2020.2.1 Maintenance Release

mweinelt, 2021-04-06:
I just backported lots of module bumps to v2020.2.x and it is probably time to do a maintenance relase anyway, given we got at least batman-adv, openssl and wolfssl updates.

mweinelt, 2021-04-12:
```
1552a15d contrib/actions: drop custom sources.list
b31a57f0 contrib/actions: drop version pin on libncurses-dev
ef3b7fe6 modules: update packages
28c5890a modules: update OpenWrt
8474fae2 gluon-core: remove unneeded KERNEL_L2TP_IP config symbol
1e9c3dea modules: update Gluon packages
2304f5c4 ramips-mt7620: remove factory images for Archer C20 v1
7f836c33 modules: update OpenWrt routing packages
240614a7 modules: update OpenWrt
10f585ed modules: update OpenWrt
366fb0ac modules: update OpenWrt
520155ec modules: update OpenWrt
6f33ff7c modules: update OpenWrt
8b8ba9fb gluon-respondd: fix count_stations() arguments
882fbab6 libgluonutil: add missing gluonutil_get_primary_domain() prototype
```

blocktrron, 2021-04-12:
- Archer C50 v4 WiFi FIX
 - RT-AC51 LED fix
 - dnsmasq fix
 - openssl 1.1.1k fix 
 - Kernel 4.14.224
 - batman-adv update from v2021.0
 - mark toggle input on EX6150 as a switch (fixes potential boots to failsafe)
 - Archer C20 v1 factory image removed

Adorfer, 2021-04-13:
having this commit run on about 20 nodes in 4 different L2domains since a week: Looks OK, nothing special observed.

mweinelt, 2021-04-14:
https://gluon.readthedocs.io/en/v2020.2.x/releases/v2020.2.3.html

PR #2202 (2021-04-14): docs: Add v2020.2.3 release notes
Closes #2194

https://github.com/freifunk-gluon/gluon/wiki/Release-Process
closed 2021-04-14 by mweinelt