freifunk-gluon/gluon #2198: Netgear r7800 misses kmod-usb-storage

smoe, 2021-04-09:
The BOINC package needs extra USB storage to function. This works nicely on the OpenWrt snapshots I tried so far, but my BOINC backport to 19.07 for my gluon router I cannot get to run because of kmod-usb-storage missing for my Netgear Nighthawk r7800. This looks like an upstream issue to me since the decision to include the package is device-specific, but you may want to direct me on where to direct any such patch.

What I was expecting was to have kmod-usb-storage available as a module, such that I could
```
# opkg install kmod-usb-storage_4.14.209-1_arm_cortex-a15_neon-vfpv4.ipk
Installing kmod-usb-storage (4.14.209-1) to root...
Configuring kmod-usb-storage.
```
to see /dev/sda as in 
```
# ls -l /dev/sda
brw-------    1 root     root        8,   0 Apr 10 00:11 /dev/sda
```
but by default, the USB stick is found and displayed via lsusb, but no offered as a device. For other platforms USB storage is available, though.

I am not exactly sure about what is the right thing to do about it. Since this is all close to the kernel, I presume that the module is best offered by the same individual who also compiles the kernel? I have hence prepared
```
diff --git a/target/linux/ipq806x/image/Makefile b/target/linux/ipq806x/image/Makefile
index e1eb090de3..30ff6d3c0f 100644
--- a/openwrt/target/linux/ipq806x/image/Makefile
+++ b/openwrt/target/linux/ipq806x/image/Makefile
@@ -191,6 +191,7 @@ define Device/netgear_r7800
        SUPPORTED_DEVICES += r7800
        DEVICE_TITLE := Netgear Nighthawk X4S R7800
        DEVICE_PACKAGES := ath10k-firmware-qca9984-ct
+       DEVICE_PACKAGE += kmod-usb-core kmod-usb2 kmod-usb3 kmod-usb-storage
 endef
 TARGET_DEVICES += netgear_r7800
```
and also did the same to a linksys wrt3200acm.

The USB drive was offered as a device immediately upon the installation of kmod-usb-storage.

I could then install cfdisk to repartition, the e2fstools to give me a filesystem, except that /proc/filesystems then informed me that there is no kmod-fs-ext4 to explain the error upon mounting the device. I then analogously helped myself with the installation of kmod-fs-ext4 but this was again only available via my own build, not to be downloaded.

Is there a way to ease the process for adding an USB stick to gluon? Or is this for every community to provide independently? 

Many thanks!
Steffen

rotanid, 2021-04-13:
you can add packages to devices or targets much easier using the site.mk file, see for example our site.mk : https://github.com/tecff/site-ffa/blob/experimental/site.mk
if this helps, please close your issue, at least i don't know what to do else here...

smoe, 2021-04-14:
Thank you for the pointer, @rotanid. I will do so, but maybe we could continue to discuss on this then closed issue what OpenWrt should change about it - they should know what machine has USB ports and which ones have not and then come up with these packages built upfront, right?

closed 2021-04-14 by smoe

rotanid, 2021-04-14:
@smoe openwrt images contain usb for this device, of course, but not the storage kmod.
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ipq806x/Makefile;h=69d3aa1e569d79a9436bf14035063986cc12c5e4;hb=HEAD#l21

smoe, 2021-04-15:
@rotanid If the philsophy is that any USB device should optionally be allowed to serve some storage purposes - then this would be a patch against OpenWrt, right? Should we then not synchronize any such patch with your site.mk? This would then live as a gluon-patch against OpenWrt and as a pull request for OpenWrt.