freifunk-gluon/gluon #2212: let "make all" call "make config" and "make update" automatically

SvenRoederer, 2021-05-05:
Is there a specific reason why the make target "all" only depends on target "config"? Why not having "config" depending on "update"?
This way a simple "make all" call will be all that needs to be done for a correctly setup build.

SvenRoederer, 2021-05-05:
For the Freifunk-Berlin firmware I can checkout, do my "site config", call make and take a coffee or so.

blocktrron, 2021-05-05:
AFAIR the reason was that invoking `update` requires internet access while this is not necessarily the case when having cloned all modules and cached all downloads.

SvenRoederer, 2021-05-06:
ok, seems a valid argument.

NeoRaider, 2021-05-06:
`make update` doesn't require internet access when all referenced modules are up-to-date, but there lots of situations where an implicit `make update` is not desirable:
- `make update` takes some time, and it needs to be run only once and not for each target
- `make update` will revert modules to their original commits after manual checkouts, which is undesirable during development

SvenRoederer, 2021-05-06:
But is this really working? I assume in most cases when the modules files changes the upstream repo has some changes to it's package version. This in turn will again trigger downloads at OpenWrt buildtime. So it might end up in a 50-50 chance that you don't need internet to build.

closed 2021-05-06 by SvenRoederer