freifunk-gluon/gluon #2225: TP-LINK C50 version 4.2 incompatible with version 4

smoe, 2021-05-20:
One of the local Freifunkers just emailed that a C50 v4.2 would allow the firmware for version 4 to be flashed, but then not work - without yet knowing what "not work" truly means. I'll chase this up.

On page https://openwrt.org/toh/tp-link/archer-c50 there is no notion of a version 4.2 to be in existence but I got an independent confirmation for that device to indeed exist on page https://www.conrad.de/de/p/tp-link-archer-c50-eu-4-2-wlan-router-2-4-ghz-5-ghz-1-2-gbit-s-1612024.html 

What is now the right thing for me/us/that local Freifunker to do? Would you have a postal address of someone who is interested to have a look if that device can be supported? Other ideas?

Many thanks!
Steffen

blocktrron, 2021-05-25:
Please first clarify what "Does not work" means exactly.

Version 4.2 uses the same firmware as version 4, so the boards should be identical (there is also a v5 using the same board as v3/v4). The recently introduces v6 uses a different 5 GHz radio and is yet unsupported.

NeoRaider, 2022-01-23:
No reply from reporter, closing. Please reopen when there is more information.

closed 2022-01-23 by NeoRaider

pure-skeleton, 2023-09-08:
i have tried installing version Archer C50(EU)_V4_180313 from tp_link website for my archer C50  router that works but router freezes, i have tried tftp installer with openwrt seems to fix solution but tp_link default farmware does not boot when i try to re install