freifunk-gluon/gluon #2232: Feature Request, gluon-autoupdater: Add Gluon version to user agent

citronalco, 2021-06-09:
Communities sign their firmware, but people come and go, and so do firmware signatures change while years pass by.

This leads to the situation that devices using older firmware can't directly update to the most current firmware because they don't accept the new signatures. To solve this the device has to be upgraded step-by-step, e.g. from v2016.1 to v2018.2 to the current one, whatever the signature "chain" allows.

Currently the autoupdater of all Gluon releases since 2017 reports itself simply with "Gluon Autoupdater (using libuclient)" as user agent.
If it would send e.g. "Gluon Autoupdater (using libuclient) v2019.1-2+3" or similar as user agent, it would be possible to configure the webserver to point it automatically to the next firmware release it is able to update to.

NeoRaider, 2021-06-09:
I like the idea, but is the Gluon version really the relevant information? Signature keys are defined in the site configuration, so the GLUON_RELEASE defined by the site seems more useful to me.

Rather than extending the user agent, how about adding custom HTTP headers for both? Something like
```
X-Gluon-Version: ...
X-Gluon-Release: ...
```

citronalco, 2021-06-09:
Sure, this would also be ok!

oakey-dev, 2021-06-09:
I think this is especially important for breaking changes between gluon versions in regard to devices that have been offline for a longer period of time.

We just recently had two devices in the network again, which hadn't been online for quite some time and hence the config upgrade would not be supported (to for e.g. v2021.1).

citronalco, 2021-06-21:
Created a PR for this: https://github.com/freifunk-gluon/packages/pull/246

closed 2021-07-16 by NeoRaider