freifunk-gluon/gluon #2233: Social issues

CodeFetch, 2021-06-09:
Since 6 years I've been commenting and contributing on the Gluon development. Unfortunately I must say that it is an uninviting environment due to the following reasons:

- Maintainers do not help much (especially newcomers) with getting their pull requests merged due to a lack of explanations. 
- Instead maintainers tend to criticize in superficial, non-constructive ways (often ignoring objections).
- Many good ideas in e.g. increasing usability were being abandoned by the PR submitters due to frustration.
- In the case where the submitters were persistent enough to submit, it often took years for their PRs to be merged.
- Non-technical issues were often not taken seriously.
- Usability has a rather low priority.
- Decisions are being made in openly meritocratic fashions whereby seemingly only the number of commits count and not their overall quality or technical background.
- The code is getting more and more complex without functionality gain for the users (e.g. rewriting Lua to C) which is an increasing entry barrier.

These are my subjective perceptions, but I'm pretty sure that I'm not the only one who sees this.

yanosz, 2021-06-10:
Sketched some guidelines a while ago: https://pad.freifunk.net/p/gluon-proposal - feel free to pick this up.

awlx, 2021-06-11:
@CodeFetch I agree that contributing to gluon is a bit of a pita. I always refrained from contributions to gluon because it was too complicated, or we simply had to touch too much code for a simple change ... so @freifunkMUC keeps a large portion of patches ... which is not how I like things to be as you know.

I really like to contribute upstream and push projects forward. But for me gluon always looks like the big monolith I fear to touch upstream, because I have to patch so much stuff to just get a minor change to work.

closed 2021-09-06 by mweinelt