freifunk-gluon/gluon #2239: sending SIGSEGV to respondd for invalid read access from 00000000

kevin-olbrich, 2021-06-13:
### Bug report

**What is the problem?**
Respondd crashes and does not provide stats.
```
[   69.704909] do_page_fault(): sending SIGSEGV to respondd for invalid read access from 00000000
[   69.714175] epc = 77f37e48 in libc.so[77ec4000+97000]
[   69.719757] ra  = 77dccb41 in gluon-mesh-vpn-core.so[77dcc000+10000]
```
Same configuration on gluon-v2020.2.2 works fine.

**What is the expected behaviour?**
Respondd should be able to start and answer to queries.

**Gluon Version:**
https://github.com/freifunk-gluon/gluon/tree/v2021.1

**Site Configuration:**
```
GLUON_FEATURES := \
	autoupdater \
	ebtables-filter-multicast \
	ebtables-filter-ra-dhcp \
	ebtables-limit-arp \
	ebtables-source-filter \
	mesh-batman-adv-15 \
	mesh-vpn-tunneldigger \
	scheduled-domain-switch \
	authorized-keys \
	lock-password \
	node-info \
	radvd \
	radv-filterd \
	respondd

GLUON_SITE_PACKAGES := \
	iwinfo \
	lldpd \
	iptables-mod-conntrack-extra \
	zram-swap \
	gluon-wifistack-quickfix \
	gluon-wifistack-reboot \
	gluon-wifistack-nodesync \
	respondd-module-airtime \
	respondd-module-wifimeshrate \
	gluon-ssid-changer

```

**Custom patches:**
None

mweinelt, 2021-06-13:
On which targets does this happen? We have upgraded to x86_64 machines and one ar71xx-generic device to v2021.1 and I don't see these segfaults.

kevin-olbrich, 2021-06-13:
> On which targets does this happen? We have upgraded to x86_64 machines and one ar71xx-generic device to v2021.1 and I don't see these segfaults.

**ar71xx-generic**

```
system type             : Qualcomm Atheros QCA956X ver 1 rev 0
machine                 : Ubiquiti UniFi-AC-PRO/MESH-PRO
processor               : 0
cpu model               : MIPS 74Kc V5.0
```

kevin-olbrich, 2021-06-13:
> Looks like you are using a custom respondd proivder: https://github.com/VfN-NRW/gluon-packages/tree/master/respondd-module-wifimeshrate

Do you think that could be an issue here (is see no link to gluon-mesh-vpn-core.so)? I will then build a new version without it.

NeoRaider, 2021-06-13:
I had a short look at the new *gluon-mesh-vpn-core* respondd provider, and there are a few places that don't handle all errors correctly. Please try the following patch:
```diff
--- a/package/gluon-mesh-vpn-core/src/respondd.c
+++ b/package/gluon-mesh-vpn-core/src/respondd.c
@@ -60,11 +60,11 @@ static struct json_object * get_bandwidth_limit(void) {
 	enabled = true;
 
 	const char *egress_str = uci_lookup_option_string(ctx, s, "limit_egress");
-	if (strcmp(egress_str, "-"))
+	if (egress_str && strcmp(egress_str, "-"))
 		egress = atoi(egress_str);
 
 	const char *ingress_str = uci_lookup_option_string(ctx, s, "limit_ingress");
-	if (strcmp(ingress_str, "-"))
+	if (ingress_str && strcmp(ingress_str, "-"))
 		ingress = atoi(ingress_str);
 
 	if (egress >= 0)
```

collimas, 2021-06-14:
I tried this patch, now respondd keeps running, but it seems it doesn't transmit data.
The node still seems to be offline on the map.

NeoRaider, 2021-06-14:
Do you get any output when running the commands
```
gluon-neighbour-info -r nodeinfo
gluon-neighbour-info -r statistics
gluon-neighbour-info -r neighbours
```
on the node?

How does the full command line of respondd look?
```
xargs -0 < /proc/$(pidof respondd)/cmdline
```

collimas, 2021-06-14:
Here we go:

root@FreifunkLippe:~# gluon-neighbour-info -r nodeinfo
root@FreifunkLippe:~# gluon-neighbour-info -r statistics
root@FreifunkLippe:~# gluon-neighbour-info -r neighbours
root@FreifunkLippe:~# xargs -0 < /proc/$(pidof respondd)/cmdline
 board=TL-WDR4300  console=ttyS0,115200 rootfstype=squashfs noinitrd

NeoRaider, 2021-06-14:
`pidof respondd` didn't print anything (so you got the kernel cmdline instead of the process cmdline), respondd is still not running at all...

kevin-olbrich, 2021-06-14:
> I had a short look at the new _gluon-mesh-vpn-core_ respondd provider, and there are a few places that don't handle all errors correctly. Please try the following patch:

Same crash.

> I tried this patch, now respondd keeps running, but it seems it doesn't transmit data.

Can not confirm this. Change has no effect for me.

> root@example:~# gluon-neighbour-info -r nodeinfo
> -ash: gluon-neighbour-info: not found

> root@example:~# xargs -0 < /proc/$(pidof respondd)/cmdline
 board=UBNT-UF-AC-PRO mtdparts=spi0.0:384k(u-boot)ro,64k(u-boot-env)ro,7744k(firmware),7744k(ubnt-airos)ro,128k(bs),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs noinitrd

> `pidof respondd` didn't print anything (so you got the kernel cmdline instead of the process cmdline), respondd is still not running at all...

At least on my node, respondd still crashes so no process.

NeoRaider, 2021-06-14:
Okay, let's start debugging this properly:
- Add `procd_set_param limits core="unlimited"` between `procd_open_instance` and `procd_close_instance` in `/etc/init.d/gluon-respondd` (in `package/gluon-respondd/files`, if you want to change this before image build)
- When respondd crashes, you should get a core file in `/tmp`. Please send the core file, as well as `/usr/bin/respondd`, `/lib/libc.so` and `/usr/lib/respondd/gluon-mesh-vpn-core.so`.

collimas, 2021-06-14:
Here's the debug information:

https://hidrive.ionos.com/share/tnf7m40ign

NeoRaider, 2021-06-14:
Oookay, two more commands to try:
```
lua -e 'print(require("gluon.mesh-vpn").enabled())'
lua -e 'name, _ = require("gluon.mesh-vpn").get_active_provider(); print(name)'
```

collimas, 2021-06-15:
`root@FreifunkLippe:~# lua -e 'print(require("gluon.mesh-vpn").enabled())'
true
root@FreifunkLippe:~# lua -e 'name, _ = require("gluon.mesh-vpn").get_active_pro
vider(); print(name)'
lua: /usr/lib/lua/gluon/mesh-vpn.lua:14: module 'gluon.mesh-vpn.provider.tunneld    igger' not found:
        no field package.preload['gluon.mesh-vpn.provider.tunneldigger']
        no file './gluon/mesh-vpn/provider/tunneldigger.lua'
        no file '/usr/share/lua/gluon/mesh-vpn/provider/tunneldigger.lua'
        no file '/usr/share/lua/gluon/mesh-vpn/provider/tunneldigger/init.lua'
        no file '/usr/lib/lua/gluon/mesh-vpn/provider/tunneldigger.lua'
        no file '/usr/lib/lua/gluon/mesh-vpn/provider/tunneldigger/init.lua'
        no file './gluon/mesh-vpn/provider/tunneldigger.so'
        no file '/usr/lib/lua/gluon/mesh-vpn/provider/tunneldigger.so'
        no file '/usr/lib/lua/loadall.so'
        no file './gluon.so'
        no file '/usr/lib/lua/gluon.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        /usr/lib/lua/gluon/mesh-vpn.lua:14: in function 'get_provider'
        /usr/lib/lua/gluon/mesh-vpn.lua:25: in function 'get_active_provider'
        (command line):1: in main chunk
        [C]: ?
root@FreifunkLippe:~#
`

collimas, 2021-06-15:
If you'd like to get access to the node for further testing, you can send your public key to me.
mb@freifunk-lippe.de

blocktrron, 2021-06-15:
@collimas can you provide me with a link to your site configuration?

collimas, 2021-06-15:
Not pushed yet, but the site config is similar.

https://github.com/freifunklippe/site-ffl

AiyionPrime, 2021-06-15:
> The new status page has problems, too.

blocktrrons PR resolves them.

AiyionPrime, 2021-06-15:
> Not pushed yet, but the site config is similar.
> 
> https://github.com/freifunklippe/site-ffl

Does the error occur in both of your site branches? (Does it matter which we build?)

collimas, 2021-06-15:
I only tested with the "lip"-branch.

AiyionPrime, 2021-06-15:
Mhm, using the siteconf from your master I run into an issue regarding gluon-banner during the build.
It does differ quite a lot from the one posted in the first comment.

Using the [above](https://github.com/freifunk-gluon/gluon/issues/2239#issue-919779553) pasted site.conf in your master throws these:
```
Configuration failed:
 * unable to enable package 'gluon-wifistack-reboot'
 * unable to enable package 'gluon-wifistack-quickfix'
 * unable to enable package 'respondd-module-wifimeshrate'
 * unable to enable package 'gluon-wifistack-nodesync'
```

Would it be possible to store the used site.conf somewhere public?

collimas, 2021-06-15:
Use my HiDrive Share above in this post.

What are these mentioned packages? I don't use them in my config.
Is there a config change necessary? I searched for things like that in the documentation but didn't find any.

AiyionPrime, 2021-06-15:
(And the site.mk respectively)

AiyionPrime, 2021-06-15:
Those are the packages OP mentioned in the first comment.

collimas, 2021-06-15:
Ah, I see. Ok, not using them in my config.

collimas, 2021-06-15:
My config still runs with Gluon 2020.x.x, the issues came with 2021.1.

AiyionPrime, 2021-06-15:
One more to go...
The packages `ffho-ap-timer` and `ffho-web-ap-timer` indicate a modified `modules`-file. Mind to share?

AiyionPrime, 2021-06-15:
> My config still runs with Gluon 2020.x.x, the issues came with 2021.1.

I don't follow. Is the config you provided not the one used to produce the issue, but just any old config you had lying around?

collimas, 2021-06-15:
Now you can find my modules file in the share.

collimas, 2021-06-15:
> 
> 
> > My config still runs with Gluon 2020.x.x, the issues came with 2021.1.
> 
> I don't follow. Is the config you provided not the one used to produce the issue, but just any old config you had lying around?

It is exactly the config I used. Just wanted to say it had no problems with older Gluon releases.

collimas, 2021-06-15:
Just pushed the actual config to the repo: https://github.com/freifunklippe/site-ffl/tree/master

NeoRaider, 2021-06-15:
> ```
> root@FreifunkLippe:~# lua -e 'print(require("gluon.mesh-vpn").enabled())'
> true
> root@FreifunkLippe:~# lua -e 'name, _ = require("gluon.mesh-vpn").get_active_pro
> vider(); print(name)'
> lua: [...]

Thank you, it seems that the new generic VPN handling code wasn't tested with Tunneldigger even once...

#2240 fixes the broken error handling in the respondd provider, #2241 should fix the underlying error.

AiyionPrime, 2021-06-15:
I compiled both PRs with the given site.
#2241 is working, one can now get past the configmode.
https://map.freifunk-lippe.de/map/#!/en/map/b44bd6224ebc

This is my first experience with tunneldigger; it appears as if there's way more broken.
Not sure, what does not work due to upstream issues.

I think the two permission hickups during gluon-reconfigure happen due to packages fflip added:

```console
# gluon-reconfigure
[...]
Configuring: 550-rsk-config
/usr/bin/gluon-reconfigure: line 9: ./550-rsk-config: Permission denied
[...]
Configuring: 888-button-bind
/usr/bin/gluon-reconfigure: line 9: ./888-button-bind: Permission denied
[...]
```

The mesh-vpn code appears to return properly now:
```console
# lua -e 'print(require("gluon.mesh-vpn").enabled())'
true
# lua -e 'name, _ = require("gluon.mesh-vpn").get_active_provider(); print(name)'
tunneldigger
```

The router appears to be reachable now, though it wasn't the first two times I connected it:
http://[2001:bf7:1350:101:b64b:d6ff:fe22:4ebc]

However, respondd is running:

```console
# xargs -0 < /proc/$(pidof respondd)/cmdline
/usr/bin/respondd -d /usr/lib/respondd -p 1001 -g ff02::2:1001 -i mesh0 -i mesh1 -i mesh-vpn -g ff05::2:1001 -i br-client -t 10
```

and provides meaningful output on all paths.

```console
# gluon-neighbour-info -r nodeinfo
{"software":{"autoupdater":{"branch":"stable","enabled":false},"batman-adv":{"version":"openwrt-2019.2-12","compat":15},"firmware":{"base":"gluon-v2021.1-9-gb83e5eec","release":"1.5.0-20210615"}},"network":{"addresses":["2001:bf7:1350:101:b64b:d6ff:fe22:4ebc","fe80::b64b:d6ff:fe22:4ebc"],"mesh":{"bat0":{"interfaces":{"wireless":["b6:4b:d6:22:4e:c0","9a:62:78:2a:dd:59"],"tunnel":["9a:62:78:2a:dd:5f"],"other":["9a:62:78:2a:dd:5b"]}}},"mesh_vpn":{"bandwidth_limit":{"enabled":false},"provider":"tunneldigger","enabled":true},"mac":"b4:4b:d6:22:4e:bc"},"location":{"latitude":52.38393700,"longitude":9.72246900},"owner":{"contact":"freifunk@aiyionpri.me"},"system":{"site_code":"fflip","domain_code":"d1","primary_domain_code":"d1"},"node_id":"b44bd6224ebc","hostname":"aiyion_tunneldigger_test","hardware":{"model":"Cudy WR1000","nproc":1}}
# gluon-neighbour-info -r statistics
{"clients":{"total":1,"wifi":0,"wifi24":0,"wifi5":0,"owe":0,"owe24":0,"owe5":0},"traffic":{"rx":{"packets":17266,"bytes":2378938},"tx":{"packets":1948,"bytes":225158,"dropped":831},"forward":{"packets":0,"bytes":0},"mgmt_rx":{"packets":7854,"bytes":1682308},"mgmt_tx":{"packets":19894,"bytes":5396716}},"gateway":"02:ca:ff:ee:01:04","gateway_nexthop":"02:ca:ff:ee:01:04","node_id":"b44bd6224ebc","time":1623770747,"rootfs_usage":0.1003,"memory":{"total":59620,"free":26380,"available":21272,"buffers":2876,"cached":9428},"stat":{"cpu":{"user":7139,"nice":0,"system":9810,"idle":128108,"iowait":0,"irq":0,"softirq":4061},"intr":635591,"ctxt":660975,"processes":6275,"softirq":670748},"uptime":1491.62,"idletime":1281.08,"loadavg":0.05,"processes":{"running":1,"total":61}}
# gluon-neighbour-info -r neighbours
{"batadv":{"9a:62:78:2a:dd:5f":{"neighbours":{"02:ca:ff:ee:01:04":{"tq":217,"lastseen":7.1699999999999999,"best":true}}}},"node_id":"b44bd6224ebc","wifi":{}}
```

Well, while I'm writing this, I lost the mesh-vpn connection again.
Maybe someone else can confirm this is regular behavior?

collimas, 2021-06-15:
Don't worry about the rsk-package permisson problem. Still need to fix it, had no time until today.

NeoRaider, 2021-06-15:
> Well, while I'm writing this, I lost the mesh-vpn connection again.
> Maybe someone else can confirm this is regular behavior?

I see two potential causes:
- 98a1c196ed8d76c0449ed85f291b4fac1ace9467 might have broken bandwidth limit settings. Please post `/etc/config/tunneldigger` and `/etc/config/simple-tc`
- It might be a tunneldigger upstream issue, as we updated to a newer version with v2021.1: https://github.com/freifunk-gluon/packages/commit/95c805c863cd0818f0e5196d3b7bde6860fd2861

AiyionPrime, 2021-06-15:
>     * Please post `/etc/config/tunneldigger` and `/etc/config/simple-tc`

Sure:

```console
# cat /etc/config/tunneldigger

config broker 'mesh_vpn'
	option uuid 'b44bd6224ebc'
	option group 'gluon-mesh-vpn'
	option broker_selection 'usage'
	option bind_interface 'br-wan'
	option interface 'mesh-vpn'
	option enabled '1'
	list address 'gate01.freifunk-lippe.de:20001'
	list address 'gate02.freifunk-lippe.de:20001'
	list address 'gate17.freifunk-lippe.de:20001'

# cat /etc/config/simple-tc
# Example config

config interface 'example'
       option enabled '0'
       option ifname 'eth0'
       option limit_egress '1000' # 1000 Kbit/s
       option limit_ingress '5000' # 5000 Kbit/s
```

AiyionPrime, 2021-06-15:
I'll keep the node at least until the online-checker is resolved; maybe I stumble across anything useful.

blocktrron, 2021-06-15:
@NeoRaider Thanks for fixing the issue with the respondd provider. I must have messed up that when renaming `method` to `provider` there.

collimas, 2021-06-16:
Thanks @ All, for your fast work! Now everything seems to work as expected.

AiyionPrime, 2021-06-17:
@kevin-olbrich The symptom got handled in #2240 and the deeper problem got fixed with the merge of #2241.
If this is indead a resolved issue, might you close this?

kevin-olbrich, 2021-06-17:
> @kevin-olbrich The symptom got handled in #2240 and the deeper problem got fixed with the merge of #2241.
> If this is indead a resolved issue, might you close this?

I'm currently preparing a new build. I will report back soon.

kevin-olbrich, 2021-06-17:
I can confirm this is working!

kevin-olbrich, 2021-06-17:
Thanks everyone for helping and fixing this issue!

closed 2021-06-17 by kevin-olbrich

RalfJung, 2021-06-20:
I'm also seeing this issue in our tunneldigger network with v2021.1, and can confirm that master works fine.

Will this be backported to the 2021.1 branch?

blocktrron, 2021-06-20:
@RalfJung I'm on it, expect a service release next week.

RalfJung, 2021-06-20:
Awesome, thanks a lot :-)