freifunk-gluon/gluon #2243: popen3 in gluon

AiyionPrime, 2021-06-17:
Recently I wrote some code, that will both need both the contents of the standard I/O streams, as well as the return value of an executed process.

in #gluon I got some references using fork() and pexec(), like we already do in
https://github.com/freifunk-gluon/gluon/blob/a2be901803469d378d6469551a65c20f744aefff/package/gluon-web-admin/luasrc/lib/gluon/config-mode/model/admin/remote.lua#L80
and https://github.com/freifunk-gluon/gluon/blob/069c8b2035c53b2235c716ec7c640e892351f5b2/package/gluon-web-admin/luasrc/lib/gluon/config-mode/controller/admin/upgrade.lua#L39

Furthermore:
> Might make sense to add a helper to gluon.util

_Originally posted by @NeoRaider  in hackint#gluon on 09.06.21 at 18:47:51_

I'd therefore like to add a helper function `popen3(path, ...)`,
that returns a newly forked pid, as well as the three streams on success,
or nil and and possibly an error message, if wanted.

NeoRaider, 2021-06-17:
Should stderr be piped as well, or inherited from the parent process? We can do whatever makes the most sense for your usecase for the initial version; when we need something different, we can add some flags to configure the behaviour.

AiyionPrime, 2021-06-17:
I intend to execute likely crashing code, which will produce stderr messages, that I'd very much like to ignore.
but given stderr as pipe one can simple raise the contents again, if desired (not by me iirc), I think.

NeoRaider, 2021-06-18:
Only use a pipe when you want to process the output, otherwise it just wastes memory (or may even block the process when you don't read it and the pipe buffer is full). Either inherit stderr from the parent process to show the errors, or set it to `/dev/null`.

AiyionPrime, 2021-06-18:
For my usecase it's enough to ignore the errors, as they're expected.
I'd set it to `/dev/null` and strip it to `popen2()`.

AiyionPrime, 2021-08-03:
Thanks, merged!

_Originally posted by @NeoRaider in https://github.com/freifunk-gluon/gluon/issues/2244#issuecomment-880815084_

closed 2021-08-03 by AiyionPrime