freifunk-gluon/gluon #2256: Statuspage-error on some TP-Link WR1043v1 with gluon-v2021.1

tackin, 2021-06-30:
With TP-Lnik WR1043v1-nodes on gluon-v2021.1-1-g0f9a6334+  the call for the statuspage randomly ends with a:
```
XML-Verarbeitungsfehler: nicht wohlgeformt
Adresse: http://[2001:bf7:fc0f:1:76ea:3aff:fee4:d1da]/cgi-bin/status
Zeile Nr. 7, Spalte 12:
 		<title><!DOCTYPE html>
```
curl tells:
```
The called action terminated with an exception:
/usr/lib/lua/gluon/web/template.lua:43: Failed to execute template &#39;status-page&#39;.
A runtime error occurred: [string &#34;/lib/gluon/status-page/view/status-page.htm...&#34;]:90: attempt to index global &#39;nodeinfo&#39; (a nil value)</pre>

```
Seems as if nodeinfo sometimes is not defined because of a timeout of  a "gluon-neighbour-info"-call.

Usually we see this only on small and older nodes-hardware with a higher load. But this one was deeply relaxed.

mweinelt, 2021-06-30:
Question is if we want to relax the timeout a bit, or if that has downsides.

https://github.com/freifunk-gluon/gluon/blob/ab4c99842a18a88d9752a30b41ba9611d8d52622/package/gluon-status-page/luasrc/lib/gluon/status-page/controller/status-page.lua#L64

grische, 2021-07-02:
I was about to report a similar bug on my wr841n v13, but those might be related.

Whenever the router boots without WiFi (disabled via the physical button), I am unable to find any status-page/provider processes:
```
root@test:~# ps  | grep status-page/provider
30986 root      1216 S    grep status-page/provider
```

That, in turn, causes the gluon-neighbour-info to return error code 1 and print no output
```
root@test:~# gluon-neighbour-info -d ::1 -p 1001 -t 60 -r nodeinfo
root@test:~# echo $?
1
```

In comparison to a working machine, I can see plenty of status-page/provider processes:
```
root@itworkshere:~# ps  | grep "status-page/provider"
10408 root       704 S    sse-multiplex exec /lib/gluon/status-page/providers/stations 'mesh0'
10414 root      1328 S    /lib/gluon/status-page/providers/stations mesh0
10415 root       704 S    sse-multiplex exec /lib/gluon/status-page/providers/stations 'mesh1'
10416 root       704 S    /usr/sbin/sse-multiplex exec /lib/gluon/status-page/providers/neighbours-batadv
10422 root      1328 S    /lib/gluon/status-page/providers/stations mesh1
19920 root      1328 S    /lib/gluon/status-page/providers/stations mesh0
...
```

Where is this status page provider started? How can I continue debugging here?

---

There is actually a second bug overlapping with the above, causing the browser error message to be so cryptic at first. The HTML is malformed and contains [half an HTML page which ends where the nodeinfo call is](https://github.com/freifunk-gluon/gluon/blob/20f665b16dca0201915b61a924347a821b18bd85/package/gluon-status-page/files/lib/gluon/status-page/view/status-page.html#L91-L97) and a second (well-formed) HTML page after that, containing the error message (probably [this one](https://github.com/freifunk-gluon/gluon/blob/a2be901803469d378d6469551a65c20f744aefff/package/gluon-status-page/files/lib/gluon/status-page/view/layout.html#L4-L24)). The embedding of both together cannot be rendered of course.

But I would open a second bug report for this if there are no objections.

blocktrron, 2021-07-03:
@mweinelt et al

Can you open a PR for increasing the timeout? So we can evaluate on a common base whether or not it improves our situation.

@grische

This issue is now tracked as #2260 

Please try 19381a23, it should fix the segfault. https://github.com/freifunk-gluon/packages/commit/825aa0c093d6c0b4f81a95cd2320331a5b5adae6#commitcomment-53000911 is still valid

Please try 19381a23, it should fix the segfault. https://github.com/freifunk-gluon/packages/commit/825aa0c093d6c0b4f81a95cd2320331a5b5adae6#commitcomment-53000911 is still valid

grische, 2021-07-04:
@blocktrron The problem with the missing nodeinfo is fixed with the most recent commit. I was also able to confirm that it was all due to a crashing respondd.

commit 4b0017f3 (2021-07-03): gluon-status-page: relax nodeinfo query timeout

It was found that a one second timeout for nodeinfo data may be too low,
so that when a node is otherwise occupied that timeout may be reached
too often.

The nodeinfo query response is also vital to the status-page base
template, so that when it times out, the site will be turned in a broken
state, that it cannot recover from.

Fixes: #2256
PR #2262 (2021-07-12): gluon-status-page: relax nodeinfo query timeout
It was found that a one second timeout for nodeinfo data may be too low,
so that when a node is otherwise occupied that timeout may be reached
too often.

The nodeinfo query response is also vital to the status-page base
template, so that when it times out, the site will be turned in a broken
state, that it cannot recover from.

Fixes: #2256
commit 76185e3a (2021-07-12): gluon-status-page: relax nodeinfo query timeout (#2262)

It was found that a one second timeout for nodeinfo data may be too low,
so that when a node is otherwise occupied that timeout may be reached
too often.

The nodeinfo query response is also vital to the status-page base
template, so that when it times out, the site will be turned in a broken
state, that it cannot recover from.

Fixes: #2256
commit 8d51a72f (2022-02-23): gluon-status-page: relax nodeinfo query timeout (#2262)

It was found that a one second timeout for nodeinfo data may be too low,
so that when a node is otherwise occupied that timeout may be reached
too often.

The nodeinfo query response is also vital to the status-page base
template, so that when it times out, the site will be turned in a broken
state, that it cannot recover from.

Fixes: #2256
(cherry picked from commit 76185e3a2a28936afb97afe0e3090fa655340565)
closed 2021-07-12 by NeoRaider