freifunk-gluon/gluon #2260: respondd crashes on sunxi-cortexa7 caused by airtime package

goligo, 2021-07-03:
Hello,

we are building an image for NanoPi R1 (sunxi-cortexa7). This was working fine in the past, but with the current master branch we found that respondd crashes

```
Sat Jul  3 13:26:36 2021 daemon.info procd: Instance gluon-respondd::instance1 s in a crash loop 6 crashes, 54 seconds since last crash
```

Backtrace is

```
(gdb) bt full
#0  0xb6ee83fc in json_object_object_add_ex (jso=jso@entry=0x0, key=key@entry=0xb6eb63f7 "phy", val=0xb6ee3d50, opts=opts@entry=0)
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/json-c-0.15/json_object.c:594
        existing_value = 0x0
        existing_entry = <optimized out>
        hash = <optimized out>
#1  0xb6ee84c4 in json_object_object_add (jso=jso@entry=0x0, key=key@entry=0xb6eb63f7 "phy", val=<optimized out>)
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/json-c-0.15/json_object.c:622
No locals.
#2  0xb6eb60e0 in respondd_provider_statistics () at respondd.c:31
        freeptr = <optimized out>
        ok = <optimized out>
        newest_element_index = <optimized out>
        last = 0x0
        result = 0x26f50
        wireless = 0xb6f00270
        ifaces = 0xb6ee3dc0
#3  0x0001260c in eval_providers (providers=0xb6f7c200) at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:357
        ret = 0xb6ee3e20
        ret = <optimized out>
#4  single_request (type=0x0, type@entry=0x26669 "statistics")
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:386
        key = {key = 0x26669 "statistics", data = 0x0}
        entry = 0xb6f79278
        r = 0xb6f79da0
        ret = <optimized out>
#5  0x00014908 in multi_request (types=0x26660 "nodeinfo")
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:411
        sub = <optimized out>
        ret = 0x26770
        type = 0x26669 "statistics"
        saveptr = 0x26674 "neighbours"
        ret = <optimized out>
        type = <optimized out>
        saveptr = <optimized out>
        sub = <optimized out>
#6  handle_request (compress=<synthetic pointer>, request=0x2665c "GET nodeinfo")
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:436
No locals.
#7  serve_request (task=task@entry=0x26630, sock=sock@entry=3)
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:493
        compress = true
        result = <optimized out>
#8  0x00011bf8 in main (argc=<optimized out>, argv=<optimized out>)
    at /home/malte/Git/ff-site-ffm/gluon-build/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/respondd-1/respondd.c:764
        task = 0x26630
        one = 1
        sock = <optimized out>
        server_addr = {sin6_family = 10, sin6_port = 59651, sin6_flowinfo = 0, sin6_addr = {__in6_union = {__s6_addr = '\000' <repeats 15 times>, __s6_addr16 = {0, 0, 0, 0, 0, 0, 0, 
                0}, __s6_addr32 = {0, 0, 0, 0}}}, sin6_scope_id = 0}
        mgroup_addr = {__in6_union = {__s6_addr = "\377\005", '\000' <repeats 11 times>, "\002\020\001", __s6_addr16 = {1535, 0, 0, 0, 0, 0, 512, 272}, __s6_addr32 = {1535, 0, 0, 
              17826304}}}
        endptr = 0xbefd9fb4 ""
        groups = 0x265f0
        c = <optimized out>
        schedule = {length = 0, list_head = 0x0}
```

As it is caused by the airtime package and there only was one change in this package in April, I assume the issue is caused by 

https://github.com/freifunk-gluon/packages/commit/8d53ff54e562ddb2ed8397781dd78edc76f6ff38

[respondd.1625311473.5213.11.zip](https://github.com/freifunk-gluon/gluon/files/6758318/respondd.1625311473.5213.11.zip)

Regards,
Malte

blocktrron, 2021-07-03:
Please try 19381a23, it should fix the segfault. https://github.com/freifunk-gluon/packages/commit/825aa0c093d6c0b4f81a95cd2320331a5b5adae6#commitcomment-53000911 is still valid

goligo, 2021-07-04:
Yes, segfault does no longer happen, after including this fix.

blocktrron, 2021-07-05:
Picked to v2021.1.x, will be included in the next service release on stable.

Thanks for reporting this issue!

closed 2021-07-05 by blocktrron