freifunk-gluon/gluon #2261: Missing dependencies

darkdragon-001, 2021-07-03:
### Bug report

**What is the problem?**

Warnings when running `make update` within [Docker container](https://github.com/freifunk-gluon/gluon/blob/master/contrib/Dockerfile) according to your [documentation](https://gluon.readthedocs.io/en/latest/user/getting_started.html#building-the-images) [duplicates removed]:

```
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist

WARNING: No feed for package 'luci-base' found
WARNING: No feed for package 'luci-lib-nixio' found
WARNING: No feed for package 'luci-lib-json' found
WARNING: No feed for package 'luci-lib-httpclient' found
WARNING: No feed for package 'luci-compat' found
```

When running `make`, I get more warnings about missing dependencies:
```
WARNING: Makefile 'package/feeds/routing/bird1-ipv4-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/bird1-ipv6-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/packages/dynapoint/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-json', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-httpclient', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-cjdns/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-cjdns/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
```

**What is the expected behaviour?**

All dependencies are installed successfully.

**Gluon Version:**

[9f4fd1d](https://github.com/GoliathLabs/gluon/commit/9f4fd1d651b9dfa7c091737d3b75c107e0fd0590)

**Site Configuration:**

[stable](https://github.com/freifunkMUC/site-ffm/commit/5a1b1f78f44b0d699dc0a1ab90815d6511bcba51)

**Custom patches:**

Only target `kirkwood-generic` for device `linksys-viper` has been added.

NeoRaider, 2021-07-03:
The first set of warnings is an OpenWrt base issue, it happens because OpenWrt base has some (optional) dependencies on the *packages* feed. There is no way to mark a dependency as "optional" (meaning it is only required for certain configurations), so the build system warns about them.

The other warnings are caused by dependencies of the *packages* and *routing* feeds on *luci*, which is not included in Gluon anymore.

While it might be possible to improve the behaviour of the feed handling in OpenWrt, I don't think it has a high priority - all these warnings are cosmetic and can be safely ignored.

blocktrron, 2021-07-03:
There is no issue here, these warnings can be ignored:

```
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
```

These are for packages in core, which have optional dependencies on packages from the packages feed. As the packages feed is not yet installed at this stage, you are seeing these warnings.

--------

```
WARNING: No feed for package 'luci-base' found
WARNING: No feed for package 'luci-lib-nixio' found
WARNING: No feed for package 'luci-lib-json' found
WARNING: No feed for package 'luci-lib-httpclient' found
WARNING: No feed for package 'luci-compat' found
```

```
WARNING: Makefile 'package/feeds/routing/bird1-ipv4-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/bird1-ipv6-openwrt/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/packages/dynapoint/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-json', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-bmx6/Makefile' has a dependency on 'luci-lib-httpclient', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-cjdns/Makefile' has a dependency on 'luci-compat', which does not exist
WARNING: Makefile 'package/feeds/routing/luci-app-cjdns/Makefile' has a dependency on 'luci-base', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
WARNING: Makefile 'package/feeds/packages/prometheus-node-exporter-lua/Makefile' has a dependency on 'luci-lib-nixio', which does not exist
```

These are because packages from the packages feed depend on the LuCI feed, which Gluon does not consume. Unless you want to build these into your Gluon image they can be safely ignored.

darkdragon-001, 2021-07-03:
Thanks for your info.

> These are because packages from the packages feed depend on the LuCI feed, which Gluon does not consume. Unless you want to build these into your Gluon image they can be safely ignored.

Since Gluon has its own web UI, LuCI is not needed, right?

blocktrron, 2021-07-03:
gluon-web is based upon LuCI 2's predecessor, so you are correct.

darkdragon-001, 2021-07-03:
Thanks for the insights.

closed 2021-07-03 by darkdragon-001

Byggvir, 2023-07-13:
I find cosmetic warning extremely irritating. 

Everyone how as a problem will star with the first warnings. They should be handled.

udaravima, 2025-01-19:
While 

> ### Bug report
> **What is the problem?**
>
> Warnings when running `make update` within [Docker container](https://github.com/freifunk-gluon/gluon/blob/master/contrib/Dockerfile) according to your [...]

You can enable some of those libraries from make menuconfig and there libraries 
I hope others also will be there