freifunk-gluon/gluon #2277: ramips-mt76x8: build waits for user input

belzebub40k, 2021-07-22:
### Bug report

**What is the problem?**

Build hangs and waits for user input when building for `ramips-mt76x8`.
```
make -C /home/admin/clones/gluon/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-5.4.128 KCFLAGS="-fmacro-prefix-map=/home/admin/clones/gluon/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl" HOSTCFLAGS="-O2 -I/home/admin/clones/gluon/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Apr 12 05:50:02 2018" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/admin/clones/gluon/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/admin/clones/gluon/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.128 CC="mipsel-openwrt-linux-musl-gcc" modules
make[6]: Entering directory '/home/admin/clones/gluon/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/linux-5.4.128'
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
net/sched/Kconfig:45: warning: menuconfig statement without prompt
*
* Restart config...
*
*
* Pulse-Width Modulation (PWM) Support
*
Pulse-Width Modulation (PWM) Support (PWM) [Y/n/?] y
  Freescale FlexTimer Module (FTM) PWM support (PWM_FSL_FTM) [N/m/y/?] n
  MediaTek PWM support (PWM_MEDIATEK) [N/m/y/?] (NEW) 

```

**What is the expected behaviour?**
The build process should run without further user interaction.

**Gluon Version:**
master (v2021.1-87-g338a831a)

**Site Configuration:**
https://github.com/freifunk-mwu/site-ffmwu/tree/restruct

**Custom patches:**
None

belzebub40k, 2021-07-22:
Also happens on rockchip-armv8. Updated first post.

NeoRaider, 2021-07-22:
@blocktrron was hunting down a few of these lately, however it is unclear to me how so many new ones turned up at once. We need to find a way to catch these in our CI...

mweinelt, 2021-07-22:
This requires some guess work. My first one would be, that they closed stdin in the shell they're giving us?

https://superuser.com/questions/813472/how-do-i-close-stdin-in-a-shell-script
https://stackoverflow.com/questions/30968734/reopen-stdout-and-stderr-after-closing-them

NeoRaider, 2021-07-22:
stdin is probably connected to `/dev/null` in the CI.

I don't see a good way to detect missing config symbols with current OpenWrt:
- Build works fine without `V=s`
- Build works fine with `</dev/null`
I believe that OpenWrt's behaviour changed at some point, and there were versions where the build without `V=s` would fail for missing symbols, but I might be wrong.

In any case, it seems that OpenWrt doesn't care too much about missing config symbols anymore, seeing how many targets have them. Maybe instead of fixing the missing symbols, we should just add a `</dev/null` at the right place in the OpenWrt build system, so things don't break with `V=s`?

blocktrron, 2021-07-30:
I'm working on a way to reproducibly trigger the failure regardloss of the presence of stdin. See my branch https://github.com/blocktrron/gluon/tree/pr-kconfig-missing

commit 596c6314 (2021-07-30): modules: update OpenWrt

a205de5594 ramips: mt76x8: add missing config symbol
8abe67d6d2 x86: move Kconfig symbol to common config
2e1a5a4353 generic: add missing Kconfig symbol

Fixes #2277
closed 2021-07-30 by blocktrron