freifunk-gluon/gluon #2281: Add support for CPE510 v3.20

micw, 2021-07-26:
### Bug report

**What is the problem?**
TP-Link CPE510 v3.20 is not supported

**What is the expected behaviour?**
TP-Link CPE510 v3.20 should be supported ^^

The master is based on OpenWRT 21.x which runs perfectly on this device. I'd be very happy if you could add support to gluon. I own such a device and can do some testing.

**Gluon Version:**
Master

**Site Configuration:**
none

**Custom patches:**
none

micw, 2021-07-26:
I have just added the line
```
device('tp-link-cpe510-v3', 'tplink_cpe510-v3')
```
to  targets/ath79-generic - that resulted in a working image. What else is needed to add the support?

nalxnet, 2021-07-28:
> What else is needed to add the support?

Working through the [Device Integration checklist](https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist) and opening a PR when you're done.

ambassador86, 2021-07-28:
Hello @micw , you could use https://github.com/freifunk-gluon/gluon/pull/2275 as blueprint.

micw, 2021-07-28:
Thank you. Most things I can fill out myself but with others I'd need some help:

```
- [ ] must be flashable from vendor firmware
  - [ ] webinterface
  - [ ] tftp
  - [ ] other: <specify>
```
I initially flashed from vendor firmware, so get this checkmark

```
- [ ] must support upgrade mechanism
  - [ ] must have working sysupgrade
    - [ ] must keep/forget configuration (if applicable)
      *think `sysupgrade [-n]` or `firstboot`*
  - [ ] must have working autoupdate
    *usually means: gluon profile name must match image name*
```
I re-flashed with sysupgrade including resetting the config. Is this sufficient?

```
- [ ] reset/wps/phone button must return device into config mode
```
Works

```
- [ ] primary mac should match address on device label (or packaging) (https://gluon.readthedocs.io/en/latest/dev/hardware.html#notes)
```
The mac printed on the device is the same as the one of eth0. Nevertheless, gluon configured a br-lan that has a completely different mac. this one is used on the LAN port for DHCP. So is that a [x] or not?

```
- wired network
  - [ ] should support all network ports on the device
  - [ ] must have correct port assignment (WAN/LAN)
```
Only one port that is used as LAN by default

```
- wifi (if applicable)
  - [ ] association with AP must be possible on all radios
  - [ ] association with 802.11s mesh must be working on all radios 
  - [ ] ap/mesh mode must work in parallel on all radios
```
There is only one radio. I have no mesh here, so impossible to test for me :-(

```
- led mapping
```
I'll check.

```
- outdoor devices only
  - [ ] added board name to `is_outdoor_device` function in `package/gluon-core/luasrc/usr/lib/lua/gluon/platform.lua`
```
I can add this in PR.

micw, 2021-08-11:
Support is there. V2.x follows soon.

closed 2021-08-11 by micw