freifunk-gluon/gluon #2286: RFC: using `$(PKG_NAME)` in Makefile of gluon packages

AiyionPrime, 2021-08-05:
The common style of other Gluon packages is not to use `$(PKG_NAME)`.

_Originally posted by @NeoRaider in https://github.com/freifunk-gluon/gluon/pull/2245#discussion_r682939879_

Why is that and if there is no strong reason why not change all to $(PKG_NAME) for convenience?

_Originally posted by @CodeFetch in https://github.com/freifunk-gluon/gluon/pull/2245#discussion_r683121065_

NeoRaider, 2021-08-07:
PKG_NAME defines the build directory name (usually the same as the source package name), while the other entries refer to binary package names. These don't need to be the same, and one source package can build multiple binary packages.

More importantly, this is not only the common style of Gluon, but for all of OpenWrt, including the package feeds I'm aware of.

Personally, I don't see why using PKG_NAME would be more convenient in this case, but this is likely a matter of taste.

AiyionPrime, 2022-01-14:
Thanks for the comment, he had about half a year to comment on this on his own.
My request was fulfilled; I'll close this.

closed 2022-01-14 by AiyionPrime