freifunk-gluon/gluon #2295: [Feature request] single_as_lan as a config option

micw, 2021-08-11:
Hello,

I'm using a TP-Link CPE510 outdoor device which has only one ethernet port. One is connected to the internet (having WAN on the ethernet port). The other is somewhere ot there, having a long-distance connection to the internet-connected one. This one should provide "LAN" on the single ethernet so that another node can be connected to it.

At the moment, I'd need to maintain 2 different site configs (including update URLs for both) which is not straightforward. I'm looking for a way to define "single_as_lan" as a device config option (that persists an auto-upgrade).

Is this possible or could we make it possible somehow?

Adorfer, 2021-10-15:
Please explain 
1)  what do you try to accomplish with that proposed feature that is not possible either by
a) disable "mesh_vpn" and enable "mesh_on_wan" or 
b) swap wan/lan in the /etc/config/network

for my understanding (e.g. if you want client-lan on the eth) the swap of lan/wan should be fully sufficient.

micw, 2021-10-15:
Swapping lan/wan _is_ actually what I want. I can do this in the siteconf which requires maintenace of 2 builds. Or I can do it manually in /etc/config/network which is not persistent. So my idea was to allow it via config page so that it can be set in a persistant way per-device.

As far as I understand, "mesh_on_wan" is a completely different setting. It does not expose the inner network on the WAN interface nor does it allow remote management on the WAN port.

maurerle, 2023-08-28:
This issue can be closed, as #2688 is merged which allows to set such behavior through the config mode in a update persistent way.

closed 2023-09-10 by neocturne