freifunk-gluon/gluon #2317: Feature: Individual AP/Client Wifi connections for mesh

micw, 2021-09-22:
Hello,

I try to find a solution for the following use case. We use some CPE510 which works pretty well with gluon. Unfortunately we cannot use it for outdoor meshing because the allowed outdoor frequencies require "Dynamic Frequency Selection" which does not work with ad-hoc networks.

A workaround is to use the CPEs with original firmware (or OpenWRT), build an Access-Point<->Client connection and use a different device with Meah-On-Lan for Freifunk.

My idea is to add an option to the router settings to configure beween "Ad-Hoc", "AP" and "Client" for the mesh interface - best would be if it's individually for 2.4 and 5 GHz. So for 5G, an outdoor frequency could be used in AP/Client mode.

yanosz, 2021-09-22:
Good one!

... have been thinking about this for a while, now. batman-adv is LLC agnostic, so there's actually no need to exclude certain modes. I'd guess that the real challange here is to derive a suitable ux. Either a user or an algorithm has to pick modes in favor. Based on the topology of a mesh, on link is more relevant than another.

As a first step, using a webUI to manual configure links is the a feasible approach.

micw, 2021-09-23:
I'd make it user-configurable and only in "advanced". Maybe directly in combination with the "outdoor mode" settings.

```
[x] outdoor mode -> Use a frequency with radar detection
(*) Disable Ad-Hoc Wifi for 5 Ghz Mesh
( ) Run as Wifi-AP for 5 Ghz Mesh
( ) Run as Wifi-Client for 5 Ghz Mesh

+ May be a channel selection
```

I did a test yesterday which looks promising:
* Manually changed channels to 104 on 2 routers
* changed /etc/config/wireless to 'ap' mode on one router
* changed /etc/config/wireless to 'sta' mode on the other router 

The other router with 'sta' mode showed some errors, after some googling I found the hint to install wpa_supplicant. Not sure if it's possible to use some different wpad package instead.

After installinf wpa_supplicant, the 'sta' routher successfully connected the mesh network to the AP router.

Both are showing as meshed in meshviewer:

[image]

yanosz, 2021-09-23:
thx for the details - not sure about the config-mode, though.

IMHO, one needs to check and associate the network while running. However, in config-mode, there's no mesh.
Having the config-mode while running (i.e. seperate wps-pushbutton-network) is an interesting option, but this is out-of-scope for this ticket, IMHO.

Rebooting the router for checking and adjusting configuration can be a frustrating experience, I think.

What I was thinking of:
* [ ] Have a shell script doing adjustments to /etc/config/wireless (as you described), maybe in a user-friendly manner. 
* [ ] Document how to adjust and and check the association using SSH
* [ ] Show those networks on the status page and submit them via respondd for mapping
* [ ] Persist these settings across reboots (related: #2159)

IMHO, this can be done in a additional package (that could also depend on packages needed for WPA).

micw, 2021-09-23:
@yanosz I did not mean to run the mesh in config mode, this was a response to "As a first step, using a webUI to configure links manually is a feasible approach."

So my idea is to have an extra setting in config mode to override the default mesh behaviour for 5ghz outdoor. Currently wifi mesh is simply disabled. With the new option, wifi mesh could either be disabled, AccessPoint or STAtion.

We also should have the option to override the SSID for AP/Station. Image a 3-hop 5G route:

````
[A]<---5Ghz--->[B]<---cable-->[C]<---5Ghz--->[D]
````

So assume A is configured as AP, B as client. The nearby C (mountet to the same location but different direction) is connected with cable to B. C is configured as AP, D as client.

We would not want to connect B to C via Wifi (because C has the same SSID as A and is nearer). That's why we also need configurable SSID for that feature.

yanosz, 2021-09-23:
> @yanosz I did not mean to run the mesh in config mode, this was a response to "As a first step, using a webUI to configure links manually is a feasible approach."
>
> So my idea is to have an extra [...]

Ah . sorry, my comment was misleading.
I was thinking about a WebUI assisting users manually configuring links using SSH. Sorry, I should have made this clear.
 
> We also should have the option to override the SSID for AP/Station. Image a 3-hop 5G route:
>
> ```
> [A]<---5Ghz--->[B]<---cable-->[C]<---5Ghz--->[D]
> ```
>
> So assume A is configured as AP, B as client. [...]

Hmmm... setting a suitable SSID is fine - however, one could also think about defining the assocation based on the AP's MAC address. Both have pro and cons; probably, its worth to revisit this idea later.

I wonder, how to go one with this ticket. Probably, this needs some feedback from the maintainers.  IMHO, it'd be a good idea to document your changes in the first place. I could think if including a guide in gluon's documentation; ideally this documation should explain how to have the configuration should be stable w.r.t firmware-updates. 

When creating a package bundling the shell script and taking care of status page and respondd to, one could either go for a community package or a package included in gluon. Personally, I'd go for the latter, because this package does not address use-cases that are only relevant for a limited number of communites - but I don't really mind.

NeoRaider, 2021-09-23:
Gluon's network configuration is already too complex, so I'm against AP/client meshing support in Gluon, especially if it requires any kind of user configuration.

As mentioned on IRC, I'd rather like to see DFS support for 11s, which is being worked on in upstream hostapd (although I'm not up-to-date on the current status). (Also, you're talking about ad-hoc networks in your proposal, which is not quite right - Gluon doesn't use Adhoc mode anymore, but only supports 11s as the layer below batadv)

yanosz, 2021-09-23:
> Gluon's network configuration is already too complex, so I'm against AP/client meshing support in Gluon, especially if it requires any kind of user configuration.

Ok, let's have a seperate package, then.
@micw Do you feel comfortable prototyping one with your changes?

micw, 2021-09-23:
> @micw Do you feel comfortable prototyping one with your changes?

Not really. I'm very new to gluon, so I need some hints where to start. I'd first document what needs to be changed to make this work. Some initial questions:

Can a package change or add things in config mode?
Can a package change the way, /etc/config/wireless is generated?
Can a package require wpad-mini and avoid that hostapd-mini is installed (which conflicts with wpad-mini)?

If that all is possible, I'm confident that it can be implemented as a package.

yanosz, 2021-09-24:
> > @micw Do you feel comfortable prototyping one with your changes?
> 
> Not really. I'm very new to gluon, so I need some hints where to start. I'd first document what needs to be changed to make this work. Some initial questions:
> 
> Can a package change or add things in config mode?

Yes, that's possible. A while ago, I created a draft for consenting to automatic updates. It adds functionality to the config-mode but works as a seperate package. For example, see
https://github.com/freifunk-gluon/gluon/commit/68ad18df77fb3eb9814d07114b8f7fbe4147b365

> Can a package change the way, `/etc/config/wireless` is generated?

The usual way is to register a shell-script which is called whenever /etc/config/wireless is generated.
I while ago I implemented setting a password for links that way for gluon-wlan-encryption-psk:

https://github.com/freifunk-gluon/gluon/commit/badbe68a1c2e21184f1fd36d4d4801f5411bcf89 - file is: `package/gluon-wlan-encryption-psk/luasrc/lib/gluon/upgrade/390-gluon-wlan-encryption-psk`

However, I wouldn't go for just modifying `/etc/config/wireless` directly based on user-input. Most probably, its more helpful to descripe all user-defined links in a file called `/etc/config/gluon-mesh-links`. This way, `/etc/config/wireless` could be generated whenever needed (i.e. after an automatic update). 

> Can a package require wpad-mini and avoid that hostapd-mini is installed (which conflicts with wpad-mini)?

That's the most tricky part, I think. In general, I did similar things when writing gluon-wlan-encryption-psk. However, this definition is applied, when a firmware image is built. 

I'm not sure, whether a package for ap/sta mesh links will be included in many community firmware images. 
I can also think of providing a package that is installed manually by people interested in hacking and tinkering. That way, one would have to deal with hostapd-mini being present in the firmware. For this use-case, one could either document how to solve the conflict using opkg or one could write tooling (shell scripts?) for installing wpad-mini when needed.

 > If that all is possible, I'm confident that it can be implemented as a package.

Just give it a try I'd suggest ;-).

yanosz, 2021-10-17:
@micw would it be fine to close this issue? It looks like this feature won't be implemented in gluon's core anytime soon. Hence, having an open ticket für freifunk-gluon/gluon doesn't look helpful to me.