freifunk-gluon/gluon #2333: Re-Add Devices from ar71xx-ath79 transition?

micw, 2021-11-20:
Hello,

it seems that after the ar71xx-ath79 transition (https://openwrt.org/docs/techref/targets/ath79#ar71xx-ath79_transition) some devices got lost. An example are the Archer C7 devices which are still fully supported by OpenWRT (just with the different target). Should these devices be added again without further tests or is it required to re-test each single device again?

Kind regards,
Michael.

NeoRaider, 2021-11-20:
A lot of devices will need some kind of config migration when upgrading from ar71xx to ath79 (possibly by deleting and regenerating /etc/config/network etc.). Our current stance is not to add devices from ar71xx to ath79 at all in Gluon (not even as BROKEN) until the migration is implemented.

AiyionPrime, 2022-02-02:
@micw as neoraiders implementation currently needs testing, you could grab a device in question and verify his migration is working as expected.
It's not much work, for an example look at the draft of #2389.

AiyionPrime, 2022-04-01:
@micw We are currently re-adding all the devices you spoke of in your original issue.
Feel free to take a look at the list and whether there's a device in "todo" you might check.

It is required to test the devices again. I hope this resolves your issue?

closed 2022-04-03 by rotanid