freifunk-gluon/gluon #2346: kernel: bridge: readd MLD querier wake-up call feature

T-X, 2021-12-21:
Unfortunately, the MLD issue with Android still hasn't improved and still persists (https://issuetracker.google.com/issues/149630944).

With the update of Gluon to OpenWrt 21.02 in https://github.com/freifunk-gluon/gluon/commit/aab2b914b8687fa the MLD "wake-up call" feature for the bridge was removed, with the removal `patches/openwrt/0015-kernel-bridge-Implement-MLD-Querier-wake-up-calls-Android-bug-workaround.patch`.

Good news is all but one kernel patch in `0015-kernel-bridge-Implement-MLD-Querier-wake-up-calls-Android-bug-workaround.patch` are part of Linux 5.4.154 and therefore part of OpenWrt 21.02 already:

**openwrt/target/linux/generic/backport-4.14/**:

* 111-bridge-simplify-ip_mc_check_igmp-and-ipv6_mc_check_m.patch
* 112-bridge-simplify-ip_mc_check_igmp-and-ipv6_mc_check_m.patch
* 113-bridge-join-all-snoopers-multicast-address.patch
* 114-bridge-Snoop-Multicast-Router-Advertisements.patch
* 115-bridge-remove-duplicated-include-from-br_multicast.c.patch
* 116-net-Fix-ip_mc_-dec-inc-_group-allocation-context.patch
* 117-net-remove-unneeded-switch-fall-through.patch
* 118-ipv6-Fix-return-value-of-ipv6_mc_may_pull-for-malfor.patch
* 119-timers-Add-a-function-to-start-reduce-a-timer.patch
* 120-bridge-Fix-a-deadlock-when-enabling-multicast-snoopi.patch

->  *10 files changed, 1908 insertions(+)* -> all now part of 5.4.154 / obsolete

---

*Still needs to be added to Gluon* and needs to be ported to Linux 5.14:

**openwrttarget/linux/generic/pending-4.14/**:

* 151-bridge-Implement-MLD-Querier-wake-up-calls-Android-b.patch

-> *1 file changed, 632 insertions(+)*

Mainly a reminder for myself, will do the update and file a PR.

blocktrron, 2022-02-09:
Just FTR: With Android 12 on my Pixel 4A this issue does not occur anymore.

This does not invalidate the effort of filing a PR, I just wanted to share this.

neocturne, 2025-03-24:
@T-X  The MLD querier wake-up call has been removed and re-added several times since this issue was opened. If we want have a tracking issue for the Android bug (until it has become irrelevant), it would probably be a good idea to open a new ticket with a better title.