freifunk-gluon/gluon #2348: Hostname not getting resolved on status page

mokkin, 2021-12-24:
### Bug report

**What is the problem?**
On some devices with activated and working mesh on LAN/WAN the hostname is not getting resolved at the status page. If this happens of wifi with a bad connection I can understand it, but here it is a cable connection.

https://freifunk-leipzig.de/map/#!/de/map/802aa8627bbd

[image]

**What is the expected behaviour?**
Hostname showing up an being clickable.

**Gluon Version:**
v2018.2.3

**Site Configuration:**
https://github.com/freifunk-leipzig/freifunk-gluon-leipzig

**Custom patches:**
none on this device

NeoRaider, 2021-12-24:
Which interface is the neighbor node using, is it mesh-on-WAN, -LAN, or a custom mesh interface?

mokkin, 2021-12-24:
This is the neighbour device: https://freifunk-leipzig.de/map/#!/de/map/30b5c20d465b
As the device has only one ethernet port, it is using mesh-on-WAN.

```
config interface 'mesh_wan'
	option ifname 'br-wan'
	option index '0'
	option proto 'gluon_wired'
	option disabled '0'
	option transitive '1'
```

AiyionPrime, 2022-01-11:
This does in my experience not only happen, if the connection is bad, but when the (neighbour-)device is under high load, too.

The device that is apparently not responding properly is a [TP-Link TL-WA850RE v1](https://openwrt.org/toh/hwdata/tp-link/tp-link_tl-wa850re_v1) which is a 4/32 device.
As I'm writing this, the load of the device floats between `1.06` and `1.26`.
I think, you would have a better freifunk-experience, if you parted with the device.

Do you have other (better) devices that are supported by recent gluon version with the same issue?

maurerle, 2023-08-28:
I can still confirm this issue with the latest release, in a installation with supported devices only:
[image]

https://map.aachen.freifunk.net/#!/en/map/18d6c790d160
Some IP-Address does not get resolved quite often, it seems like the node stops resolving before it resolved all connections..?

I can reproduce this on FFMUC router status pages as well, but it happens less frequently there (general resolving is much faster too)