freifunk-gluon/gluon #2353: difficult test situation regarding babel

AiyionPrime, 2021-12-30:
In the last year I encountered several PRs that touched code near babel.
Naturally testing such code implies building a recent gluon with a compatible site.

Usually I grab our communities nightly siteconf and tweak it.
Our community does not use babel though, which means I have neither a proper siteconf nor the infrastructure to test such self-built images against.

Taking another communities siteconf and network for testing purposes is usually the next thing I tend to.
This appears to become increasingly difficult regarding babel.

There are several communities that have some sort of gluon-babel-images in their  network, but apparently not so many with recent let alone nightly images, which I could rely on.

So if there's a community with regular gluon-babel-builds, I'd be glad to know about.

Gossip has it, gluon-babel would not even work on recent gluon commits;
(And hasn't for at least half a year or so.)

I'd really like to see that proven wrong.
So if your community may help me out on this, let me know!

lemoer, 2021-12-30:
CC: @christf

AiyionPrime, 2022-01-10:
> 12:21 <Klaus_D1eter> Ich muss einfach Mal aktuell bauen. Bin aber Recht eingespannt im Moment und mache daher keine Versprechen
12:22 <Klaus_D1eter> Magdeburg nutzt einen nicht upstream patchset wegen wireguard
12:22 <Klaus_D1eter> Mit l3 vpn

christf, 2022-12-13:
... the MD community is alive again

AiyionPrime, 2022-12-13:
@christf Testing this would be highly appreciated :)

AiyionPrime, 2022-12-14:
Looking forward to see Magdeburgs babel community thrive. Thanks again!