freifunk-gluon/gluon #2355: add aarch64 & x86 vm as kernel + blockdev devices

txt-file, 2021-12-30:
My community recently got a request for aarch64 virtual machine image. Intended usage was offloader.

I looked into [OpenWrt](https://openwrt.org/docs/guide-user/virtualization/qemu#openwrt_in_qemu_aarch64) and they provide vm images split into kernel & storage. IMHO the split is a good idea. I do not yet know how to define kernel + blockdev files. So I open it as issue.

I would like to have x86 also as split kernel + blockdev files.

NeoRaider, 2021-12-30:
Generally sysupgrade requires a combined kernel+rootfs image with partition table, it will not work with split images (as the kernel image can't even be accessed from inside the VM). Providing these images for boot tests as `extra_images` isn't a problem, but the usefulness is limited.

T-X, 2021-12-31:
For LXC/LXD support as requested in https://github.com/freifunk-gluon/gluon/issues/1583 a separate aarch64 rootfs tarball might be useful.

Ideally OpenWrt would also be able to output the portable LXD image files, which are basically a tarball of rootfs tarball + an LXD YAML config file, to ease setup. And OpenWrt sysupgrade would then know how to detect and sysupgrade such LXD images).

NeoRaider, 2021-12-31:
@T-X Please keep LXC-related discussion in #1583.

NeoRaider, 2021-12-31:
I've looked into the aarch64 VM images, and while OpenWrt does have an armvirt target, it still uses the legacy image build code, which will not work anymore since 6ba58c9b17c9e41b521d796ab76e5723ee017170. I might look into converting the target at some point (like it was done with x86 in OpenWrt 21.02), but at the moment other issues have higher priority.

neocturne, 2023-12-24:
We now have ARM v7 and v8 VM support using the `armsr` targets. Like the standard x86 images, this defaults to a combined image, so sysupgrade is supported.

Separate images like added for x86 in #2359 *may* still be useful for development, but we haven't added them for ARM so far (and I'm not sure if doing so even makes sense for a UEFI-based target)

closed 2025-03-24 by neocturne