freifunk-gluon/gluon #2375: babel: firewall issues with mesh-on-wan without vxlan

T-X, 2022-01-18:
### Bug report

#### Setup Description

* Local, virtual test setup on a laptop with 2 nodes and 2 clients:

```
client0 <-> [node0] <= mesh-on-wan => [node1] <-> client1
```
* node0, node1: x86 KVM Gluon nodes (current master, on 7427ba2280ac1)
* client0, client1: hosts in separate network namespaces
* mesh-on-WAN, no VXLAN

#### Issue

Nodes and clients can't ping each other on their "node_prefix6" and "node_client_prefix6" addresses when using mesh-on-wan without VXLAN. Nodes don't recognize each other in BABEL.

Babeld's status dump only shows a node's own routes and own client routes, but not from other nodes. tcpdump shows BABEL's hello messages, but no "I-heard-you" responses.

Calling `ip6tables -I INPUT -p udp --dport 6696 -j ACCEPT` and `ip6tables -I INPUT -p udp --dport 5523 -j ACCEPT` makes BABEL IHU messages appear and allows BABEL's routing table to fill up correctly. And allows the two nodes to ping each other and allows a node to ping both clients. While pinging between clients still is rejected by the direct node ("Destination unreachable: Port unreachable"). Enabling VXLAN makes things work as well, including pinging between the two clients.

#### Additional Information

**Gluon Version:**
* a recent Gluon master: 7427ba2280ac1

**Site Configuration:**
* https://gist.github.com/T-X/9dbab63be36782aa73c7694fced3643d
  * site-default-novxlan.conf -> not working
  * site-vxlan.conf -> working

Concerns: #2353

T-X, 2022-01-18:
What actually surprises me a bit is that mesh-on-wan with VXLAN works. When doing a `$ iptables -S | grep 6696` in my KVM setup with no wifi interface I get no results. So I'm wondering if it works by accident, if VXLAN by accident is less strict than mesh-on-wan without VXLAN (zone "wired_mesh" rejects INPUT and FORWARD, but I don't see a zone for "wired_mesh_vxlan").

There are rules "mesh_babel" and "mesh_l3roamd" for zone "mesh" (which I believe is wireless only?). But I don't see such babel/l3roamd rules for zone "wired_mesh".

NeoRaider, 2022-01-18:
With Babel, there is no way to distinguish mesh and non-mesh traffic on the same interface by the packet type, so this simply cannot work - after all, mesh traffic must never leak into the WAN network for security reasons!

Non-VXLAN wired meshing is deprecated for batadv, and was never intended to be possible with Babel at all. It is a bug in our site checkers that it is possible to build a Babel-based firmware with `vxlan = false`.

NeoRaider, 2022-01-18:
Note that the reason for the behavior you're seeing is that all mesh traffic through WAN will be filtered by the regular WAN zone by Babel (blocking most of it), while batadv uses a non-IP Ethertype which bypasses the firewall entirely.

mkg20001, 2022-01-19:
If babel doesn't explicitly needs it's own VLAN another idea might be to use a macvlan then