freifunk-gluon/gluon #2380: OpenWrt patch "wireguard-tools: allow generating private_key (v3)" is not in OpenWrt

NeoRaider, 2022-01-23:
This patch was added in #2168 (cc @lemoer)

We've been trying to reduce the number of downstream patches for a while now. Adding new patches that are not in OpenWrt is not acceptable.

AiyionPrime, 2022-01-23:
https://patchwork.ozlabs.org/project/openwrt/patch/20210915010918.83602-1-me@irrelefant.net/

Appears to be under review, still.

NeoRaider, 2022-01-23:
@AiyionPrime Thanks for the info. Might need a ping...

AiyionPrime, 2022-01-23:
I've pinged ynezz in openwrt-devel and asked him, what help he could make use of.
I'll see what comes of it.

AiyionPrime, 2022-01-26:
I sent a mail on the lsit as well, but did not heir of him either way.
Will do this on a weekly basis now.

AiyionPrime, 2022-02-08:
ynezz' has been diligent: https://patchwork.ozlabs.org/project/openwrt/patch/20210915010918.83602-1-me@irrelefant.net/

The result is https://github.com/openwrt/openwrt/commit/5406684087815d3f66df9d8318e15db7137148f5

I think this can be closed then as the patch now is part of openwrt.

blocktrron, 2022-02-08:
Looks good.

closed 2022-02-08 by blocktrron