freifunk-gluon/gluon #2384: TP-Link WBS210 v2: Ethernet interfaces are swapped

citronalco, 2022-01-27:
Checked a build of the current master (https://github.com/freifunk-gluon/gluon/commit/2924afe5624d1445fdb39a6a9f2cb8470e3458d6)
The TP-Link WBS210v2 has two LAN ports, one of them, LAN0, is needed for powering the device over PoE.
Currently eth0 and eth1 are swapped: Setup interface is on the PoE powered port "LAN0", while the WAN port is assigned to "LAN1".
It should be the other way round, otherwise two cables are needed.

NeoRaider, 2022-01-27:
It is possible that LAN and WAN are switched; the PoE port should be WAN by default, but I have no way to check that without access to the hardware. However, I'm surprised that the setup interface is on LAN - as far as I can tell, the device is correctly marked as an "outdoor device", which should move the setup interface to the WAN port.

Please post the output of `tail /lib/gluon/core/sysconfig/*_ifname`.

citronalco, 2022-01-27:
```
==> /lib/gluon/core/sysconfig/lan_ifname <==
eth1

==> /lib/gluon/core/sysconfig/setup_ifname <==
eth0

==> /lib/gluon/core/sysconfig/wan_ifname <==
eth0
```

NeoRaider, 2022-01-27:
You wrote that the setup interface is on the PoE-powered port LAN0. The same port becomes uplink/WAN after configuration, which is the intended setup for outdoor devices.

citronalco, 2022-01-27:
```
==> /lib/gluon/core/sysconfig/lan_ifname <==
eth1

==> /lib/gluon/core/sysconfig/setup_ifname <==
eth0

==> /lib/gluon/core/sysconfig/wan_ifname <==
eth0

```

Configuration on the PoE powered port. WAN is on the non-powered port.
I'd expect it either the other way round, or that both is on the PoE port.

NeoRaider, 2022-01-27:
Are you sure? `setup_ifname` and `wan_ifname` are both shown as `eth0`, so they must be the same port.

citronalco, 2022-01-27:
Rechecked, and you're right: Config is also on the non-powered port. Sorry.

NeoRaider, 2022-01-27:
Should be fixed by #2385, please test.

citronalco, 2022-01-28:
Tested, it works: Setup and WAN are now both on the PoE port. Thank you!

PR #2385 (2022-01-28): ath79-generic: swap interfaces on TP-Link WBS210 v2
Swap the interfaces so than the PoE input port LAN0 is used for WAN and
config mode, and LAN1 becomes LAN.

To this end, the code previously used for ar71xx and removed in
commit 9fdc57c175b4 ("treewide: drop ar71xx platform specific code") is
reintroduced.

Fixes #2384
commit 3ea770db (2022-01-28): ath79-generic: swap interfaces on TP-Link WBS210 v2 (#2385)

Swap the interfaces so than the PoE input port LAN0 is used for WAN and
config mode, and LAN1 becomes LAN.

To this end, the code previously used for ar71xx and removed in
commit 9fdc57c175b4 ("treewide: drop ar71xx platform specific code") is
reintroduced.

Fixes #2384
closed 2022-01-28 by NeoRaider