freifunk-gluon/gluon #2396: Long hostname breaks respondd

SmithChart, 2022-02-16:
### Bug report

**What is the problem?**
During our weekly meetup a community member reported that a node did not show up on our map. We investigated the issue and traced it back to respondd.

Steps to reproduce:
1) Flash a router with a fresh gluon installation.
2) Get yourself a really long string as a hostname, eg. using ``curl https://gluon.readthedocs.io/en/latest/ | base64  | xclip -selection c -i
``.
3) Use this string as the name of the new node in the config mode wizard.
4) After reboot we will find the following config in ``uci``:
```
system.@system[0].pretty_hostname='PCFET (.... the rest of the very long string...)
system.@system[0].hostname='PCFET0NUWVBFIGh0bWwCjxodG1sIGNsYXNzPSJ3cml0ZXItaHRtbDQiIGxhbmc9'
```
5) The node will not show up on the communities map.
6) If we query respndd on the node itself we will see the following:
```
root@PCFET0NUWVBFIGh0bWwCjxodG1sIGNsYXNzPSJ3cml0ZXItaHRtbDQiIGxhbmc9:/# gluon-neighbour-info -i lo -d ::1 -r nodeinfo
{"software":{"autoupdater":{"branch":"stable","enabled":true},"batman-adv":{"version":"openwrt-2019.2-12","compat":15},"firmware":{"base":"gluon-v2021.1.1-9-gaf0f3a13","release":"20211030"}}
,"network":{"addresses":["2a0f:b506:ff05:0:9ade:d0ff:feb4:e85c","fe80::9ade:d0ff:feb4:e85c"],"mesh":{"bat0":{"interfaces":{"wireless":["72:4c:49:0a:78:e9"],"tunnel":["72:4c:49:0a:78:ef"],"ot
her":["72:4c:49:0a:78:eb"]}}},"mesh_vpn":{"bandwidth_limit":{"enabled":false},"provider":"tunneldigger","enabled":true},"mac":"98:de:d0:b4:e8:5c"},"owner":{"contact":"Chrissi^ @ Freifunk Bra
unschweig"},"system":{"site_code":"ffnw","domain_code":"default","primary_domain_code":"default"},"node_id":"98ded0b4e85c","hostname":"PCFET0NUWVBFIGh0bWw+CjxodG1sIGNsYXNzPSJ3cml0ZXItaHRtbDQ
iIGxhbmc9ImVuIiA+Cjxo ZWFkPgogIDxtZXRhIGNoYXJzZXQ9InV0Zi04IiAvPgogIDxtZXRhIG5hbWU9InZpZXdwb3J0IiBj b250ZW50PSJ3aWR0aD1kZXZpY2Utd2lkdGgsIGluaXRpYWwtc2NhbGU9MS4wIiAvPgogIDx0aXRs ZT5XZWxjb21lIH
RvIEdsdW9uICZtZGFzaDsgR2x1b24gMjAyMS4xIGRvY3VtZW50YXRpb248L3Rp dGxlPjxsaW5rIHJlbD0ic3R5bGVzaGVldCIgaHJlZj0iX3N0YXRpYy9jc3MvdGhlbWUuY3NzIiB0 eXBlPSJ0ZXh0L2NzcyIgLz4KICAgIDxsaW5rIHJlbD0ic3R5bG
VzaGVldCIgaHJlZj0iX3N0YXRp Yy9weWdtZW50cy5jc3MiIHR5cGU9InRleHQvY3NzIiAvPgogICAgICA8bGluayByZWw9InN0eWxl c2hlZXQiIGhyZWY9Il9zdGF0aWMvY3NzL2N1c3RvbS5jc3MiIHR5cGU9InRleHQvY3NzIiAvPgog ICAgPGxpb
msgcmVsPSJjYW5vbmljYWwiIGhyZWY9Imh0dHBzOi8vZ2x1b24ucmVhZHRoZWRvY3Mu aW8vZW4vbGF0ZXN0L2luZGV4Lmh0bWwiIC8+CiAgPCEtLVtpZiBsdCBJRSA5XT4KICAgIDxzY3Jp cHQgc3JjPSJfc3RhdGljL2pzL2h0bWw1c2hpdi5taW4ua
nMiPjwvc2NyaXB0PgogIDwhW2VuZGlm XS0tPgogIDxzY3JpcHQgaWQ9ImRvY3VtZW50YXRpb25fb3B0aW9ucyIgZGF0YS11cmxfcm9vdD0i Li8iIHNyYz0iX3N0YXRpYy9kb2N1bWVudGF0aW9uX29wdGlvbnMuanMiPjwvc2NyaXB0PgogICAg ICAg
IDxzY3JpcHQgdHlwZT0idGV4dC9qYXZhc2NyaXB0IiBzcmM9Il9zdGF0aWMvanF1ZXJ5Lmpz Ij48L3NjcmlwdD4KICAgICAgICA8c2NyaXB0IHR5cGU9InRleHQvamF2YXNjcmlwdCIgc3JjPSJf c3RhdGljL3VuZGVyc2NvcmUuanMiPjwvc2NyaXB0
PgogICAgICAgIDxzY3JpcHQgdHlwZT0idGV4 dC9qYXZhc2NyaXB0IiBzcmM9Il9zdGF0aWMvZG9jdG9vbHMuanMiPjwvc2NyaXB0PgogICAgICAg IDxzY3JpcHQgdHlwZT0idGV4dC9qYXZhc2NyaXB0IiBzcmM9Il9zdGF0aWMvbGFuZ3VhZ2VfZGF0
 YS5qcyI+PC9zY3JpcHQ+CiAgICAgICAgPHNjcmlwdCBhc3luYz0iYXN5bmMiIHR5cGU9InRleHQv amF2YXNjcmlwdCIgc3JjPSJodHRwczovL2Fzc2V0cy5yZWFkdGhlZG9jcy5vcmcvc3RhdGljL2ph dmFzY3JpcHQvcmVhZHRoZWRvY3MtZG9jLWV
tYmVkLmpzIj48L3NjcmlwdD4KICAgIDxzY3JpcHQg c3JjPSJfc3RhdGljL2pzL3RoZW1lLmpzIj48L3NjcmlwdD4KICAgIDxsaW5rIHJlbD0iaW5kZXgi IHRpdGxlPSJJbmRleCIgaHJlZj0iZ2VuaW5kZXguaHRtbCIgLz4KICAgIDxsaW5rIHJlbD0
ic2Vh cmNoIiB0aXRsZT0iU2VhcmNoIiBocmVmPSJzZWFyY2guaHRtbCIgLz4KICAgIDxsaW5rIHJlbD0i bmV4dCIgdGl0bGU9IkdldHRpbmcgU3RhcnRlZCIgaHJlZj0idXNlci9nZXR0aW5nX3N0YXJ0ZWQu aHRtbCIgLz4gCgo8IS0tIFJURCBFeH
RyYSBIZWFkIC0tPgoKPGxpbmsgcmVsPSJzdHlsZXNoZWV0 IiBocmVmPSJodHRwczovL2Fzc2V0cy5yZWFkdGhlZG9jcy5vcmcvc3RhdGljL2Nzcy9yZWFkdGhl ZG9jcy1kb2MtZW1iZWQuY3NzIiB0eXBlPSJ0ZXh0L2NzcyIgLz4KCjxzY3JpcHQgdH
lwZT0iYXBw bGljYXRpb24vanNvbiIgaWQ9IlJFQURUSEVET0NTX0RBVEEiPnsiYWRfZnJlZSI6IGZhbHNlLCAi YXBpX2hvc3QiOiAiaHR0cHM6Ly9yZWFkdGhlZG9jcy5vcmciLCAiYnVpbGRfZGF0ZSI6ICIyMDIy LTAyLTE1VDE0OjUyOjIxWiIsI
CJidWlsZGVyIjogInNwaGlueCIsICJjYW5vbmljYWxfdXJsIjog bnVsbCwgImNvbW1pdCI6ICI2NWY1YTNiMCIsICJkb2Nyb290IjogIi9kb2NzLyIsICJmZWF0dXJl cyI6IHsiZG9jc2VhcmNoX2Rpc2FibGVkIjogZmFsc2V9LCAiZ2xvYmFsX2FuY
Wx5dGljc19jb2Rl IjogbnVsbCwgImxhbmd1YWdlIjogImVuIiwgInBhZ2UiOiAiaW5kZXgiLCAicHJvZ3JhbW1pbmdf bGFuZ3VhZ2UiOiAibHVhIiwgInByb2plY3QiOiAiZ2x1b24iLCAicHJveGllZF9hcGlfaG9zdCI6 ICIvXyIsICJzb3VyY2Vf
c3VmZml4IjogIi5yc3QiLCAic3VicHJvamVjdHMiOiB7fSwgInRoZW1l IjogInNwaGlueF9ydGRfdGhlbWUiLCAidXNlcl9hbmFseXRpY3NfY29kZSI6ICIiLCAidmVyc2lv biI6ICJsYXRlc3QifTwvc2NyaXB0PgoKPCEtLQpVc2luZyB0aGlzIHZh
cmlhYmxlIGRpcmVjdGx5 IGluc3RlYWQgb2YgdXNpbmcgYEpTT04ucGFyc2VgIGlzIGRlcHJlY2F0ZWQuClRoZSBSRUFEVEhF RE9DU19EQVRBIGdsb2JhbCB2YXJpYWJsZSB3aWxsIGJlIHJlbW92ZWQgaW4gdGhlIGZ1dHVyZS4K LS0+CjxzY3JpcHQ
gdHlwZT0idGV4dC9qYXZhc2NyaXB0Ij4KUkVBRFRIRURPQ1NfREFUQSA9IEpT T04ucGFyc2UoZG9jdW1lbnQuZ2V0RWxlbWVudEJ5SWQoJ1JFQURUSEVET0NTX0RBVEEnKS5pbm5l ckhUTUwpOwo8L3NjcmlwdD4KCjxzY3JpcHQgdHlwZT0idGV4dC9
qYXZhc2NyaXB0IiBzcmM9Imh0 dHBzOi8vYXNzZXRzLnJlYWR0aGVkb2NzLm9yZy9zdGF0aWMvamF2YXNjcmlwdC9yZWFkdGhlZG9j cy1hbmFseXRpY3MuanMiIGFzeW5jPSJhc3luYyI+PC9zY3JpcHQ+Cgo8IS0tIGVuZCBSVEQgPGV4 dHJhaGVhZD
4gLS0+CjwvaGVhZD4KCjxib2R5IGNsYXNzPSJ3eS1ib2R5LWZvci1uYXYiPiAKICA8 ZGl2IGNsYXNzPSJ3eS1ncmlkLWZvci1uYXYiPgogICAgPG5hdiBkYXRhLXRvZ2dsZT0id3ktbmF2 LXNoaWZ0IiBjbGFzcz0id3ktbmF2LXNpZGUiPgogICAgIC
A8ZGl2IGNsYXNzPSJ3eS1zaWRlLXNj cm9sbCI+CiAgICAgICAgPGRpdiBjbGFzcz0id3ktc2lkZS1uYXYtc2VhcmNoIiA+CiAgICAgICAg ICAgIDxhIGhyZWY9IiMiIGNsYXNzPSJpY29uIGljb24taG9tZSI+IEdsdW9uCiAgICAgICAgICA8 L2E+C
iAgICAgICAgICAgICAgPGRpdiBjbGFzcz0idmVyc2lvbiI+CiAgICAgICAgICAgICAgICBs YXRlc3QKICAgICAgICAgICAgICA8L2Rpdj4KPGRpdiByb2xlPSJzZWFyY2giPgogIDxmb3JtIGlk PSJydGQtc2VhcmNoLWZvcm0iIGNsYXNzPSJ3eS1mb
3JtIiBhY3Rpb249InNlYXJjaC5odG1sIiBt ZXRob2Q9ImdldCI+CiAgICA8aW5wdXQgdHlwZT0idGV4dCIgbmFtZT0icSIgcGxhY2Vob2xkZXI9 IlNlYXJjaCBkb2NzIiAvPgogICAgPGlucHV0IHR5cGU9ImhpZGRlbiIgbmFtZT0iY2hlY2tfa2V5 
d29yZHMiIHZhbHVlPSJ5ZXMiIC8+CiAgICA8aW5wdXQgdHlwZT0iaGlkZGVuIiBuYW1lPSJhcmVh IiB2YWx1ZT0iZGVmYXVsdCIgLz4KICA8L2Zvcm0+CjwvZGl2PgogICAgICAgIDwvZGl2PjxkaXYg Y2xhc3M9Ind5LW1lbnUgd3ktbWVudS12ZXJ0
aWNhbCIgZGF0YS1zcHk9ImFmZml4IiByb2xlPSJu YXZpZ2F0aW9uIiBhcmlhLWxhYmVsPSJOYXZpZ2F0aW9uIG1lbnUiPgogICAgICAgICAgICAgIDxw IGNsYXNzPSJjYXB0aW9uIj48c3BhbiBjbGFzcz0iY2FwdGlvbi10ZXh0Ij5Vc2VyIERvY3Vt
ZW50 YXRpb248L3NwYW4+PC9wPgo8dWw+CjxsaSBjbGFzcz0idG9jdHJlZS1sMSI+PGEgY2xhc3M9InJl ZmVyZW5jZSBpbnRlcm5hbCIgaHJlZj0idXNlci9nZXR0aW5nX3N0YXJ0ZWQuaHRtbCI+R2V0dGlu ZyBTdGFydGVkPC9hPjwvbGk+CjxsaSB
jbGFzcz0idG9jdHJlZS1sMSI+PGEgY2xhc3M9InJlZmVy ZW5jZSBpbnRlcm5hbCIgaHJlZj0idXNlci9zaXRlLmh0bWwiPlNpdGUgY29uZmlndXJhdGlvbjwv YT48L2xpPgo8bGkgY2xhc3M9InRvY3RyZWUtbDEiPjxhIGNsYXNzPSJyZWZlcmVuY2U
gaW50ZXJu YWwiIGhyZWY9InVzZXIvc3VwcG9ydGVkX2RldmljZXMuaHRtbCI+U3VwcG9ydGVkIERldmljZXMg JmFtcDsgQXJjaGl0ZWN0dXJlczwvYT48L2xpPgo8bGkgY2xhc3M9InRvY3RyZWUtbDEiPjxhIGNs YXNzPSJyZWZlcmVuY2UgaW50ZX
JuYWwiIGhyZWY9InVzZXIveDg2Lmh0bWwiPng4NiBzdXBwb3J0 PC9hPjwvbGk+CjxsaSBjbGFzcz0idG9jdHJlZS1sMSI+PGEgY2xhc3M9InJlZmVyZW5jZSBpbnRl cm5hbCIgaHJlZj0idXNlci9mYXEuaHRtbCI+RnJlcXVlbnRseSBBc2tlZCBRdW
VzdGlvbnM8L2E+ PC9saT4KPC91bD4KPHAgY2xhc3M9ImNhcHRpb24iPjxzcGFuIGNsYXNzPSJjYXB0aW9uLXRleHQi PkZlYXR1cmVzPC9zcGFuPjwvcD4KPHVsPgo8bGkgY2xhc3M9InRvY3RyZWUtbDEiPjxhIGNsYXNz PSJyZWZlcmVuY2UgaW50Z
XJuYWwiIGhyZWY9ImZlYXR1cmVzL2NvbmZpZ21vZGUuaHRtbCI+Q29u ZmlnIE1vZGU8L2E+PC9saT4KPGxpIGNsYXNzPSJ0b2N0cmVlLWwxIj48YSBjbGFzcz0icmVmZXJl bmNlIGludGVybmFsIiBocmVmPSJmZWF0dXJlcy9hdXRvdXBkYXRlci5od
G1sIj5BdXRvdXBkYXRl cjwvYT48L2xpPgo8bGkgY2xhc3M9InRvY3RyZWUtbDEiPjxhIGNsYXNzPSJyZWZlcmVuY2UgaW50 ZXJuYWwiIGhyZWY9ImZlYXR1cmVzL3dsYW4tY29uZmlndXJhdGlvbi5odG1sIj5XTEFOIGNvbmZp Z3VyYXRpb248L2E+
PC9saT4KPGxpIGNsYXNzPSJ0b2N0cmVlLWwxIj48YSBjbGFzcz0icmVmZXJl bmNlIGludGVybmFsIiBocmVmPSJmZWF0dXJlcy9wcml2YXRlLXdsYW4uaHRtbCI+UHJpdmF0ZSBX TEFOPC9hPjwvbGk+CjxsaSBjbGFzcz0idG9jdHJlZS1sMSI+PGEg
Y2xhc3M9InJlZmVyZW5jZSBp bnRlcm5hbCIgaHJlZj0iZmVhdHVyZXMvd2lyZWQtbWVzaC5odG1sIj5XaXJlZCBtZXNoIChNZXNo LW9uLVdBTi9MQU4pPC9hPjwvbGk+CjxsaSBjbGFzcz0idG9jdHJlZS1sMSI+PGEgY2xhc3M9InJl ZmVyZW5jZSB
pbnRlcm5hbCIgaHJlZj0iZmVhdHVyZXMvZG5zLWZvcndhcmRlci5odG1sIj5ETlMg Zm9yd2FyZGVyPC9hPjwvbGk+CjxsaSBjbGFzcz0idG9jdHJlZS1sMSI+PGEgY2xhc3M9InJlZmVy ZW5jZSBpbnRlcm5hbCIgaHJlZj0iZmVhdHVyZXMvbW9uaXR
vcmluZy5odG1sIj5Ob2RlIG1vbml0 b3Jpbmc8L2E+PC9saT4KPGxpIGNsYXNzPSJ0b2N0cmVlLWwxIj48YSBjbGFzcz0icmVmZXJlbmNl IGludGVybmFsIiBocmVmPSJmZWF0dXJlcy9tdWx0aWRvbWFpbi5odG1sIj5NdWx0aWRvbWFpbiBT dXBwb3
[... cut ...] to add a upper limit at the same time.

I am not really sure how to go about this. Do you know a simple way to limit the length in the input field?
Should respondd take care of this? 

**Gluon Version:**
Tested with Freifunk Braunschweig: Gluon v2020.2.3 + Site https://gitli.stratum0.org/ffbs/ffbs-site/-/commit/f5ab2d55f7d47db2a91ab16f84503afdf88a59a7
And tested with Freifunk Nord: Gluon v2021.1.x + Site https://git.ffnw.de/ffnw-firmware/siteconf/-/compare/20210915...rc%2F20211030

Thanks!
Chris

AiyionPrime, 2022-02-17:
Could you test, whether this occurs with builds with a recent gluon? (after the merge of https://github.com/freifunk-gluon/gluon/pull/2322?)

SmithChart, 2022-02-17:
Thank! I will give it a spin and will get back to you.

AiyionPrime, 2022-02-18:
Nevertheless you and neoraider are right, the input should forbid too long inputs.

There are various length limits we could consider:

Labels in dns-names (the part between dots) are not supposed to be longer than 63 characters. This is our current limit.

NetBIOS is even stricter; I think that was the reason you could not name windows PCs longer than 15 characters.
(Have not used Windows in a while, quite possible that's no longer a restriction)

In hanover there are various routers with a descriptive name like "[UFU-FWH-A273-Mecklenheide-Buero](https://hannover.freifunk.net/karte/#/e063dab090d2)" (31 chars).

I think 15 would be too short, any other length suggestions @neoraider?

AiyionPrime, 2022-02-18:
Not sure whether there are parallels to `hostnamectl --pretty` but as an excerpt its pretty hostname does not limit the length to anything below even 400 chars:

> Note that the pretty hostname has little restrictions on the characters and length used, while the static and transient hostnames are limited to the usually accepted characters of Internet domain names, and 64 characters at maximum (the latter being a Linux limitation).

Maybe the cleanest would be to leave it untouched then, but partly revert or alter https://github.com/freifunk-gluon/gluon/commit/7e0075584de98f3fbad5694b2ceeffac5b215301#diff-1a84d4504bf1020582522dcad02ba53c32f4c261f0fa5d36280ce524049642bdR82 (I don't like that approach too much and already can hear utf-**n** fans crying.

belzebub40k, 2022-02-18:
I check the hostname length for our domains and the majority is below 30 characters but there are a few with up to 50 characters.

This is the command I used to run a quick check on our Meshviewer data.

`curl -s https://map.freifunk-mwu.de/data/nodelist.json | jq -r '.nodes[].name' | awk '{print length}' | sort -n | uniq -c`

NeoRaider, 2022-02-18:
I think one of the following options to solve this would make sense:
1. Limit the length of the node name that can be set in the config wizard
2. Limit the length of the node name that is transmitted via respondd
3. Limit both (my preference)

Note that this doesn't only apply to the node name, but to all user-defined data (like the contact field, which is currently also unlimited). A limit of 100 characters might be good enough.

SmithChart, 2022-02-20:
> e.g. our nightly: https://build.ffh.zone/job/gluon-nightly/ws/download/images/sysupgrade/

Tested the behavior on:
```
Gluon version / Site version:	v2021.1-265-gbb25f35 / vH20-34-gd6ce7f4
Firmware release:       	vH21.N20220220
```

Now ``gluon-neighbour-info -i lo -d ::1 -r nodeinfo`` returns valid JSON with the very long hostname.

AiyionPrime, 2022-02-20:
That's good to see, thanks for testing it.
Yet neoraider is right about the unneccessary length.

maurerle, 2023-08-28:
[image]

One can discuss the visualization in the meshviewer, but it works to have 126 char names, which I think is enough for all sane cases.
I think this issue can be closed..?

SmithChart, 2024-10-09:
With the long hostname the `respondd` `json`-response is now intact. (But the node does not show up on our map.)

With just 100 chars everything is fine. I'll close this now. Thanks!

closed 2024-10-09 by SmithChart