freifunk-gluon/gluon #2397: Futro S400 kernel panic: invalid opcodehhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhhh

eliba, 2022-02-18:
### Bug report

**What is the problem?**

**What is the expected behaviour?**

**Gluon Version:**

**Site Configuration:**

**Custom patches:**

NeoRaider, 2022-02-18:
The `x86-generic` image requires at least Pentium 4 or compatible. For the AMD NX the `x86-geode` image should be appropriate (and we also have an `x86-legacy` image for even less capable CPUs).

eliba, 2022-02-20:
Thanks NeoRaider, I tried both image variants, and it seems to be an even less capable CPU. It works with the `x86-legacy` image, however only when booting off a USB thumb drive. When trying to boot off the CompactFlash card, it still hangs at the line:
```
clocksource: Switched to clocksource tsc
```
This is sufficent for now and I will continue to boot from USB, since the Futro offers it as a primary boot option.

Installing a Debian 11 onto the CompactFlash card works, so I suspect there's a driver missing in gluon.

NeoRaider, 2022-02-22:
If you can find out which driver is missing (the Debian kernel log might contain a hint), I don't see an issue with including it in our x86 images by default.

eliba, 2022-02-23:
This would be awesome!
I can't really tell the missing driver's name from the log alone, so here's the full `dmesg` output. The disk in question got recognized as `sda`. I also extracted `dmidecode` and `lspci` just in case, when I had Debian installed. Let me know if you need those as well.
[dmesg.txt](https://github.com/freifunk-gluon/gluon/files/8123268/dmesg.txt)

NeoRaider, 2022-02-23:
The device uses the `pata_sis` driver. I've pushed a branch `hardware/pata_sis` based on the Gluon master that enables it. If you can't build an image yourself, I can also upload a test build.

eliba, 2022-02-23:
I'll try next week and let you know immediately. Thank you already :)

eliba, 2022-02-28:
Yessss it works perfectly!

NeoRaider, 2022-03-10:
Thanks for testing. I've pushed the change to OpenWrt master and 21.02, so with the next module update the Gluon master will get it as well.

closed 2022-03-10 by NeoRaider