freifunk-gluon/gluon #2413: ar71xx - ath79 migration progress

AiyionPrime, 2022-03-21:
@NeoRaider's recent contribution of the network regeneration got merged (thanks to all involved!)
and if I'm not mistaken, we've agreed on testing the devices that were gone again, as if we'd be adding new support for them.

That's arguably an effort, but hopefully one divided on several shoulders.

Below is a list of devices that got dropped with the target ar71xx, which hadn't been marked as tiny back then and was not within the mikrotik subtarget.
I think that would be a solid start for the migration.

I'll order them in ongoing, todo and done. Please sue `ctrl+f` to find the device you are interested in.

Testing the devices is not only flashing from stock and via sysupgrade, but sysupgrading from an old ar71xx image as well.
You do not need to build an ar71xx image for this, as hanover has them available [here](http://firmware.ffh.zone/vH21/) and will provide you with access to its VPN infrastructure. If possible let us know when you're done with the key.

Thanks for your help,
Aiyion.

**in progress**

**todo**

**done**

**to assess**

| done | subtarget | tester         | PR    | device                              |
|------|-----------|----------------|-------|-------------------------------------|
| <ul><li>- [ ] </li></ul> | generic   |                |       | 8devices-carambola2-board           |
| [ ]  | nand      |                |       | aerohive-hiveap-121                 |
| [ ]  | generic   |                |       | allnet-all0315                      |
| [ ]  | generic   |                |       | avm-fritz-box-4020                  |
| [ ]  | generic   |                |       | avm-fritz-wlan-repeater-300e        |
| [ ]  | generic   |                |       | avm-fritz-wlan-repeater-450e        |
| [ ]  | generic   |                |       | buffalo-wzr-hp-g300nh               |
| [ ]  | generic   |                |       | buffalo-wzr-hp-g300nh2              |
| [ ]  | generic   |                |       | buffalo-wzr-hp-g450h                |
| [ ]  | generic   |                |       | buffalo-wzr-hp-ag300h               |
| [ ]  | generic   |                |       | buffalo-wzr-600dhp                  |
| [ ]  | generic   |                |       | d-link-dap-1330-rev-a1              |
| [ ]  | generic   |                |       | d-link-dir-505-rev-a1               |
| [ ]  | generic   |                |       | d-link-dir-505-rev-a2               |
| [ ]  | generic   |                |       | gl-inet-6408a-v1                    |
| [ ]  | generic   |                |       | gl-inet-6416a-v1                    |
| [ ]  | generic   |                |       | gl.inet-gl-ar150                    |
| [ ]  | generic   |                |       | gl.inet-gl-ar300m                   |
| [ ]  | generic   |                |       | gl.inet-gl-ar750                    |
| [ ]  | generic   |                |       | gl.inet-gl-usb150                   |
| [ ]  | generic   |                |       | meraki-mr12 (was broken)            |
| [ ]  | generic   |                |       | meraki-mr16 (was broken)            |
| [ ]  | generic   |                |       | netgear-wndr3700                    |
| [ ]  | generic   |                |       | netgear-wndr3700v2                  |
| [ ]  | nand      |                |       | netgear-wndr3700v4                  |
| [ ]  | generic   |                |       | netgear-wndr3800                    |
| [ ]  | generic   |                |       | netgear-wndr3800chmychart           |
| [ ]  | nand      |                |       | netgear-wndr4300                    |
| [ ]  | generic   |                |       | netgear-wndrmacv2                   |
| [ ]  | generic   |                |       | netgear-wndrmac                     |
| [ ]  | generic   |                |       | netgear-wnr2200 (was broken)        |
| [ ]  | generic   |                |       | ocedo-koala                         |
| [ ]  | generic   |                |       | onion-omega (was broken)            |
| [ ]  | generic   |                |       | openmesh-a60                        |
| [ ]  | generic   |                |       | openmesh-a40                        |
| [ ]  | generic   |                |       | openmesh-mr1750                     |
| [ ]  | generic   |                |       | openmesh-mr1750v2                   |
| [ ]  | generic   |                |       | openmesh-mr600                      |
| [ ]  | generic   |                |       | openmesh-mr600v2                    |
| [ ]  | generic   |                |       | openmesh-mr900                      |
| [ ]  | generic   |                |       | openmesh-mr900v2                    |
| [ ]  | generic   |                |       | openmesh-om2p                       |
| [ ]  | generic   |                |       | openmesh-om2pv2                     |
| [ ]  | generic   |                |       | openmesh-om2pv4                     |
[... 50 lines cut ...]
| [ ]  | generic   |                |       | ubiquiti-nanostation-m-xw           |
| [ ]  | generic   |                |       | ubiquiti-nanostation-m2-xw          |
| [ ]  | generic   |                |       | ubiquiti-nanostation-m5-xw          |
| [ ]  | generic   |                |       | ubiquiti-rocket-m-xw                |
| [ ]  | generic   |                |       | ubiquiti-rocket-m2-xw               |
| [ ]  | generic   |                |       | ubiquiti-rocket-m5-xw               |
| [ ]  | generic   |                |       | ubiquiti-rocket-m-ti                |
| [ ]  | generic   |                |       | ubiquiti-rocket-m2-ti               |
| [ ]  | generic   |                |       | ubiquiti-rocket-m5-ti               |
| [ ]  | generic   |                |       | ubiquiti-unifi                      |
| [ ]  | generic   |                |       | ubiquiti-unifi-ap                   |
| [ ]  | generic   |                |       | ubiquiti-unifi-ap-lr                |
| [ ]  | generic   |                |       | ubiquiti-unifi-ap-pro               |
| [ ]  | generic   |                |       | ubiquiti-unifiap-outdoor            |
| [ ]  | generic   |                |       | ubiquiti-unifiap-outdoor+           |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-lite              |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-lite-mesh         |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-lr                |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-pro               |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-mesh              |
| [ ]  | generic   |                |       | ubiquiti-unifi-ac-mesh-pro          |
| [ ]  | generic   |                |       | wd-my-net-n600                      |
| [ ]  | generic   |                |       | wd-my-net-n750                      |
| [ ]  | generic   |                |       | zyxel-nbg6616                       |
| [ ]  | nand      |                |       | zyxel-nbg6716                       |

AiyionPrime, 2022-03-22:
@s-2 I'll add your efforts to the list above, once you drafted the PRs for the respective devices.

Regarding DIR-825 B1, it's has only 6M for firmware and has been marked as tiny before in https://github.com/freifunk-gluon/gluon/commit/cfce3ee91e17708df44331b45f7b80cb6256b458.

Feel free to try, but tiny is out of scope for this tracker, as I wrote above.

NeoRaider, 2022-03-28:
I can also take care of the tp-link-archer-c7-v2.

Dark4MD, 2022-03-28:
> I can also take care of the tp-link-archer-c7-v2.

i have it nearly finished, just waiting for a new build to finish because i was missing to add the wifi driver and if that works i post the PR.

heis2201, 2022-03-28:
Hi @AiyionPrime , I got your DM on https://forum.freifunk.net/, feel free to use my github account @heis2201.

I am happy to contribute by testing the migration process, however, I need to find some spare time and I have to figure out how to flash the OEM firmware back onto the tp-link-tl-wdr3500-v1.

Is there is chance I can brick the device to a state where recovery is impossible? Did I get this right that test migration of each device is documented in a separate issue here on github?

Cheers

rotanid, 2022-03-28:
@AiyionPrime what does "wants to test" in your list mean? the text above suggests that this person intends to create a PR, which is not the same as testing something. i don't have the time to create PRs for all the devices mentioning my name.
also, i don't have a C59 and a wr1043v5 at the moment to test, but i have a ubiquiti-unifi-ac-lr
also note, ubiquiti-unifi-ac-mesh is no separate device but an alias, as far as i know.

NeoRaider, 2022-03-28:
> also note, ubiquiti-unifi-ac-mesh is no separate device but an alias, as far as i know.

It's a separate device in ath79. As no dynamic model detection exists anymore, even completely identical devices get separate images, so each can show the correct model name.

AiyionPrime, 2022-03-29:
> Hi @AiyionPrime , I got your DM on https://forum.freifunk.net/, feel free to use my github account @heis2201.

Nice, I'll update the list accordingly.

> I am happy to contribute by testing the migration process, however, I need to find some spare time and I have to figure out how to flash the OEM firmware back onto the tp-link-tl-wdr3500-v1.
> 
> Is there is chance I can brick the device to a state where recovery is impossible?

The OpenWrt page is worth a visit, when it comes to sharing experiences with flashing in various directions.
While it should not happen, it's technically possible to brick your device.
Most devices do have some form of recovery though, which OpenWrt (and therefore gluon as well) aim not to overwrite.

If you are uncomfortable doing the testing alone, you can always ask for help in hackint#gluon, there are enough people who have done the same process before.

> Did I get this right that test migration of each device is documented in a separate issue here on github?

That I try to open a seperate PR for every device I add helps me to stay on top of what I'm doing, when I work on five different devices.
There's no rule you cannot do more that one device at once, which @ecsv impressively demonstrated a week ago :D
It just keeps things simple.

AiyionPrime, 2022-03-29:
> @AiyionPrime what does "wants to test" in your list mean? the text above suggests that this person intends to create a PR, which is not the same as testing something. i don't have the time to create [...]

The list does contain every device or alias we'd possibly intend to migrate to ath79.
That I added aliases as well just helps to see which aliases must not be forgotten at first glance. And gives ideas like "didnt these x revisions of that device all have the same hardware".

"Wants to test" is a mere reference, someone has ever announced interest in supporting the migration process for that particular device.
The list is not endless; when you have time to do a migration on your own that's great.
And when there's someone that "just" owns a device and would like to help I think that's worth a lot as well.

I currently have a break after my exams, so whoever has a device listed in "todo" above and cannot do the migration on his own, I'm happy to assist. That might not be enough in all cases, but should be sufficient for first time contributors like @heis2201.

And if you have no time to prepare the PRs, get me a list of a few devices you'd like to "just test" and I'll build the images for you as well.

heis2201, 2022-03-29:
OK, I'm happy doing some testing over the next few days. Where can I see a list of test cases? Has anyone assembled a guide or at least recommends an order of doing things. Haven't spend any time thinking about it, but right now I would say, I test the migration from ar71xx to ath79 using `sysupgrade`, and in a second test I flash the OEM image and then flash the ath79 `factory` image. More or less correct?

AiyionPrime, 2022-03-29:
The list you are looking for would be https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist .

The order depends on whether you've got stock or ar71xx firmware on your device.
But looking at the list, you'll figure that out.

Have you built gluon before? And looking at recently closed PRs, do you wnat to do the PR on your own?

heis2201, 2022-03-29:
I have never built gluon myself. 

One think I haven't understood, do we need to test stock --> gluon (ath79) or is it sufficient to do gluon ar71xx --> ath79?

I am happy to do the PR on my own.

AiyionPrime, 2022-03-29:
https://gluon.readthedocs.io/en/latest/user/getting_started.html#building-the-images

That's a good point to start from.
You'd build it locally, building gluon from scratch costs me about an hour on my 2013 laptop.
Rebuilding is faster.

Then good luck with it :)

heis2201, 2022-03-29:
Where would I find the container that is mentioned in the docs? Otherwise I would have to spin up and update a Debian VM first, which would take much longer.

Also, what the the `RELEASE`?  `v2021.1.1`? Next, what would be my site configuration? The one from my regional freifunk community?

This thing is a bit bigger than I was hoping for. 

Should I stop my support questons here, and we continue the conversation elsewhere (freifunk forum for example)?

AiyionPrime, 2022-03-29:
If you want I'm available in hackint#gluon. You could join via webchat, that should be the fastest way.

heis2201, 2022-03-29:
I have an IRC client and a matrix account, but I am afraid this would be something I would do late night, and rather spontaneous. But will see, once I get to it, I will be in touch

udoschneider, 2022-03-30:
I can offer to test a ubiquiti-unifi-ac-mesh-pro. Will be this WE earliest though.

AiyionPrime, 2022-04-01:
Kommt auf das Gerät an. Die erste Spalte der Tabelle gibt an, in welchem der beiden Subtargets das Gerät in ar71xx war.

AiyionPrime, 2022-04-01:
Die Geräte, die du erarbeiten wolltest, scheinen generic zu sein.

AiyionPrime, 2022-04-02:
The "ubiquiti-unifi-ac-lite-mesh" is not thing. Goal is 118 now.

AiyionPrime, 2022-04-03:
> I can offer to test a ubiquiti-unifi-ac-mesh-pro. Will be this WE earliest though.

__@udoschneider__ earlier in this thread

Looking at recently closed PRs, you might find enough inspiration to do the PR on your own.
Let me know, if you'd like to avoid building gluon yourself and would rather "just" flash and test the device;
in which case I'd offer to provide the PR for you.

udoschneider, 2022-04-07:
> Looking at recently closed PRs, you might find enough inspiration to do the PR on your own.
> Let me know, if you'd like to avoid building gluon yourself and would rather "just" flash and test the device;
> in which case I'd offer to provide the PR for you.
If you have a ready to flash firmware I'd happy to try this now. I'm able to build gluon using the ffmuc site but don't understand the build process enough to guarantee I'm building the correct branch.

AiyionPrime, 2022-04-07:
I will notify you, when it's built. ETA 35m.

AiyionPrime, 2022-04-07:
@udoschneider https://github.com/freifunk-gluon/gluon/pull/2462

AiyionPrime, 2022-04-13:
Getsern Abend war das [zweite Entwickler-Meeting](https://github.com/freifunk-gluon/gluon/wiki/Meeting-2022-02) diesen Jahres.

In Anlehnung an das Gespräch ergeben sich in der Liste oben zwei Änderungen.
Zum einen wurde mir der Unterschied zwischen dem Target `tiny` und der Device-Class `tiny` aufgezeigt. Mein Ziel war es, das ar71xx-Tiny-Target in diesem PR nicht zu berücksichtigen.
Die fehlenden Geräte habe ich den Listen oben hinzugefügt.
Desweiteren haben wir einen besonderen Fokus auf Geräte beschlossen, die laut Multi-Meshviewer mehr als hundert mal betrieben werden.
Diese Geräte von besonderer Dringlichkeit sollten leichter aufzutreiben sein und mehr Sinn stiften, als der Rest und sind obig **dick** markiert.
Vielen Dank allerseits.

AiyionPrime, 2022-06-14:
@rotanid closed the related PRs already, in todays Developer Meetup we agreed to dismiss the migration of xx/32 devices. That's not all of device-class tiny, but a significant ease on this issue.

That's not meant to be understood as "we don't want Gluon to support them", but merely none of of us is currently willing to put the necessary effort into these devices to get them running on a recent kernel again.

@NeoRaider suggested, someone willing could test an ath79-tiny approach. But that's not my intention to track here, as long as no one stepped up to get them working again.

sheilbronn, 2022-07-03:
> @rotanid closed the related PRs already, in todays Developer Meetup we agreed to dismiss the migration of xx/32 devices. That's not all of device-class tiny, but a significant ease on this issue. [...]

Hello @rotanid and @NeoRaider,
Just to be sure about some specific 8/32 devices, since I have several [TL-WR710N v1.2 and v2.1](https://openwrt.org/toh/tp-link/tl-wr710n) running Freifunk München and other community images.
You are saying these WR710N's won't have a future, unless somebody will develop an ath79-tiny approach, even though they support OpenWrt 21.02.3 ?
Is somebody aware of any work having started on this approach?
Thanks very much!

rotanid, 2022-07-03:
> Hello @rotanid and @NeoRaider, Just to be sure about some specific 8/32 devices, since I have several [TL-WR710N v1.2 and v2.1](https://openwrt.org/toh/tp-link/tl-wr710n) running Freifunk München [...]

Yes, like every device with less than 64M memory. we tested many things and didn't achieve a usable image.

as a sidenote, OpenWrt 21.02 is not relevant as Gluon v2022.1 will be based on OpenWrt 22.03.

> Is somebody aware of any work having started on this approach? 

no to my knowledge

blocktrron, 2022-07-03:
>  You are saying these WR710N's won't have a future, unless somebody will develop an ath79-tiny approach, even though they support OpenWrt 21.02.3 ?

What is Important to note here (As it is missing) is the fact a -tiny approach would only delay the inevitable by ~1 year. Thus nobody at the developer meetups was interested in such a non-longterm solution.

rotanid, 2022-07-07:
**LAST CALL - diese Geräte werden sonst im Release fehlen!**

es gibt an die 200 Geräte die in den Communities online sind, die genug Speicher/RAM haben aber wo uns Tester und/oder Testgeräte fehlen, z.B. folgende. (alle die öfter als 10 mal vorkommen)
Bitte dringend bei uns melden!

Ubiquiti UniFi AC MESH PRO
Ubiquiti Rocket M2
AVM Fritz!WLAN Repeater 300E
Netgear WNDR3700v4
Netgear WNDR3800
Ocedo Koala
TP-Link TL-WR2543N

rubo77, 2022-07-07:
> * p-link-cpe220-v1
> 
> * tp-link-cpe220-v1.1
> 
> * tp-link-cpe520-v1.1

Oben steht "never sold", 
Wir haben diese sehr häufig in Kiel und Nord, daher klingt die beunruhigend.
gibt es dazu schon nähere Erläuterungen irgendwo?

AiyionPrime, 2022-07-07:
> > * p-link-cpe220-v1
> > * tp-link-cpe220-v1.1
> > * tp-link-cpe520-v1.1
> 
> Oben steht "never sold", Wir haben diese sehr häufig in Kiel und Nord, daher klingt die beunruhigend. gibt es dazu schon nähere Erläuterungen irgendwo?

Wenn du wirklich irgendwo eine 520er oder 220er hast, baue ich dir alle Images, die du brauchst. Sicher dass das keine 10er sind?

rubo77, 2022-07-07:
Stimmt, verkuckt. Die Kommentare können alle wieder weg

rubo77, 2022-07-12:
@heini66 kann ein paar geräte bereitstellen, (kontaktiert ihn bitte in matrix)

ambassador86, 2022-07-13:
Wir haben in Chemnitz auch mindestens eine CPE 220. Da kann ich das mal testen.

AiyionPrime, 2022-07-13:
@ambassador86 Mhm, gerne. Aber ihr habt da nicht 'auch' eine CPE220. Ihr hättet die erste und einzige, die wir im Migrationsprozess gesehen hätten.

AiyionPrime, 2022-07-13:
> @heini66 kann ein paar geräte bereitstellen, (kontaktiert ihn bitte in matrix)

@rubo77 Wenn du ihm schon gesagt hast, dass wir das so rum machen, von mir aus.

Generell würde ich mich freuen, wenn die Leute, die wollen, dass man Ihnen bei der Migration ihrer Geräte unter die arme greift selber ein bisschen Initiative zeigen.
*hier irgendwas über Berge und Propheten einfügen**

ambassador86, 2022-07-13:
@AiyionPrime sogar im OpenWRT Wiki steht drin, dass die Dinger von Varia verkauft werden. Und die sitzen hier bei uns.

AiyionPrime, 2022-07-13:
> @AiyionPrime sogar im OpenWRT Wiki steht drin, dass die Dinger von Varia verkauft werden. Und die sitzen hier bei uns.

Wenn die sich anfinden, sehr gerne. Wollte nur vermeiden, dass du deine Zeit verschwendest, falls du auch einen Zahlendreher gehabt hättest. Bin optimistisch gespannt.

AiyionPrime, 2022-07-13:
> @heini66 kann ein paar geräte bereitstellen, (kontaktiert ihn bitte in matrix)

Ist angeschrieben.

ambassador86, 2022-07-17:
@AiyionPrime ich hab naturlich nicht richtig hingeschaut und es sind beide v3 daher kann die v1 vermutlich raus, wenn sich keine anfindet.

AiyionPrime, 2022-12-13:
As of our meetup today, we'd like to remove this effort from the roadmap, as we are mostly done and this issue is becoming more and more stale.
We do not intend to exclude the remaining missing devices from our codebase,
but would like to revoke the prominent position this issue has been taking for months now.

Thanks to everyone involved so far - and happy hunting for the remaining "treasures" to whoever finds them.

Djfe, 2023-02-06:
I could help by trying to contact the one device owner in Aachen of an "ubiquiti-unifiap-outdoor" (non-plus variant) for testing it.
It's currently listed as "wanted to test" by yourself @AiyionPrime 
Do you have access yourself or is help appreciated?

AiyionPrime, 2023-02-07:
We bought such a router and bricked it upon testing -.-'
I ordered a SOP16 clip to flashback the image I dumped before.

But until it arrived in europe, someone already tried to solder around the ground pin of the chip...
Until I've undone this help, I cannot test it again.

Unless you know someone who could unbrick the device in case the update really bricks routers, I'd be hesitant to accept help :/

I won't get to it until april due to exams.

Djfe, 2023-02-07:
that sucks 😣
I'll refrain from offering help. If there's a chance to brick, then I won't try to convince anyone to let us test new fw on their device. It doesn't make any sense to brick more devices that are in "production" and working.

techtimo, 2023-04-01:
@AiyionPrime 
I have a NBG6716 to test some images. Currently running stock ath79.

AiyionPrime, 2023-04-03:
@techtimo That sounds great.
The first step is to open a Pull-Request with changes similar to this one: https://github.com/freifunk-gluon/gluon/pull/2460/files

Clone this repo locally,
create a new branch `ath79-nbg6716`
and make in the same files at the same places in the files as linked above.
Replace `6616` with your `6716` and commit your changes with the message `ath79: (re)add ZyXel NBG6716`.
Push your changes and we can guide you through your next steps in the PR you'll open :)

Djfe, 2023-05-08:
Update, the device runs, we might open a PR at some point.
Mesh on 5GHz isn't working though so that will probably need some fixes and I don't want to create a draft PR that sits around for ages.
It could be related to https://github.com/openwrt/openwrt/issues/10784
techtimo's device has 4MiB NOR-Flash instead of 16MiB like the one that was originally added. other's have 8MiB.
NAND-Flash can be either 256 or 128MiB, [see Wiki](https://openwrt.org/toh/zyxel/nbg6716#hardware_highlights)

I'm not sure whether the art partition contains garbage though:
[nbg6716-art.zip](https://github.com/freifunk-gluon/gluon/files/11417128/nbg6716-art.zip)
Maybe someone can take a look at it to tell me

[Full dmesg log](https://pastebin.com/mBCiKMg3)

excerpt from logread when I run `wifi down&wifi up` [full log](https://pastebin.com/NmZjLpJ2)
```
Mon May  8 03:34:41 2023 kern.info kernel: [39930.224544] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Mon May  8 03:34:41 2023 kern.warn kernel: [39930.255284] ath10k_pci 0000:00:00.0: could not get mac80211 beacon, vdev_id: 0 addr: d6:9e:0b:a3:60:90
Mon May  8 03:34:41 2023 kern.info kernel: [39930.271348] ath10k_pci 0000:00:00.0: mac flush vdev 0 drop 0 queues 0x1 ar->paused: 0x0  arvif->paused: 0x0
Mon May  8 03:34:41 2023 kern.info kernel: [39930.311951] ath10k_pci 0000:00:00.0: mac flush null vif, drop 0 queues 0xffff

[...]

Mon May  8 03:37:44 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy client0) --> new PHY
Mon May  8 03:37:46 2023 kern.warn kernel: [40115.239867] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 4  peers: 96  tid: 256
Mon May  8 03:37:46 2023 kern.info kernel: [40115.255996] ath10k_pci 0000:00:00.0: wmi print 'P 97 V 4 T 371'
Mon May  8 03:37:46 2023 kern.info kernel: [40115.262392] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
Mon May  8 03:37:46 2023 kern.info kernel: [40115.270565] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 41968 iram: 50724'
Mon May  8 03:37:46 2023 kern.warn kernel: [40115.335907] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
Mon May  8 03:37:46 2023 kern.info kernel: [40115.359921] ath10k_pci 0000:00:00.0: rts threshold -1
```

AiyionPrime, 2023-05-08:
I think that counts as an upstream issue :)
Maybe you can check whether it has been addressed in OpenWrts master already and if not, offer testing/fixing of the device upstream.

Djfe, 2023-05-25:
Update 5GHz runs fine now.
The issue was literally, that this is a wave 1 device, so ath10k-ct doesn't support meshing..
I only had to add the packages variable for qca988x to use ath10k instead

This still means that there might be other variants of nbg6716 out there where it won't work
https://github.com/openwrt/openwrt/issues/10784

How do we proceed? I suggest adding the device to Gluon and then figuring out a fix for the other variant afterwards, since it requires feedback from external people which I can't rely on.
I wouldn't want to add the device as broken, but in the end it's your call.

AiyionPrime, 2023-05-25:
After reading through your comments twice, I feel like I'm still lacking information; and that's not a new problem ;)

Anyway. If you feel like a device can be migrated, open a PR and explain potential problems in the PR instead of this tracking issue.

If you fiddle with an ath10k change, do try to limit to affected models, not every model in that target.
Good luck :)

AiyionPrime, 2023-12-29:
We just talked about testing more of the listed devices above during the meet-up at the congress.
Please do not modify the checklist above manually, we've got a script for it.

If you can offer testing for a device above let us know which model via comment.

Djfe, 2023-12-29:
testing of/for what? testing for the migration or testing whether master runs fine on an already migrated device? :)

AiyionPrime, 2023-12-30:
The former.

Djfe, 2023-12-30:
you may update the hiveap 121 above. the change has been merged in May already but the posts still state
`unlikely/wontfix` in two places

Djfe, 2023-12-30:
same goes for the Zyxel NBG6716
#2913
though the solution is not ideal.
It requires a reinstall (factory image) and sysupgrades from older releases aren't possible.

Also the new ath79 build supports only up to 128mb of flash and was never tested on variants with 256mb of flash (could still be done), but I assume it should still run. (if @belzebub40k 's user is still available for testing and has such a variant of the device an addition test would be nice)

oakey-dev, 2024-01-09:
we would have the following Devices running or somewhere on reserve at Freifunk Ingolstadt:
- ubiquiti-unifiap-outdoor
- netgear-wndrmacv2 (should be here somewhere)
- 8devices-carambola2-board (we have a contact)

AiyionPrime, 2024-01-10:
Hello @oakey-dev, that's nice to hear.
Just to set expectations straight beforehand though:
We can provide you with Testimages, the testing against the checklist needs to be done near the device though.
If it does not work out at once more restrains might be in order.

I'd suggest you start with the device easiest to aquire or most interesting to you and see, whether the process suits you enough to proceed with the others.

oakey-dev, 2024-01-24:
The ubiquiti-unifiap-outdoor is mounted in a location which is rather inaccessible.
(I have access to the PoE injector that has a reset button, but it doesn't seam to be working)

AiyionPrime, 2024-01-24:
You'll need physical access to its port, its LEDs, its mac address label and all that preferably not on a pole.

GoliathLabs, 2024-03-14:
I'll probably be able to test the netgear-wndrmacv2 with Freifunk München.

[9 more replies]

closed 2025-06-27 by AiyionPrime