freifunk-gluon/gluon #2419: statuspage: duplicate meshneighbors

AiyionPrime, 2022-03-23:
### Bug report

**What is the problem?**

The statuspage shows (local) mesh neighbors of both bands twice or thrice.

[image]

The js code that updates them throws an index error, which is only a symptom.

I first encountered the problem yesterday, when I built a PR on master.

I bisected the master branch, starting in with a commit in january I think:

> d728f17b8fc848b35e93eebb8e948798a1e4440a is the first bad commit

It appears the error is target independent, I encountered it in `ath79` and reproduced it in `ramips-mt76x8`.

**What is the expected behaviour?**

The meshneighbors adresses (which exist once) should be shown once.

**Gluon Version:**

The error is present in the current master.
Our local nightly version does contain it, if somebody wants to see it in action:

Thanks to snuggels for providing a test device:
http://[2001:678:978:222:8ede:f9ff:fec3:7cde]/cgi-bin/status

**Site Configuration:**

The site config does not have an effect, tested with our current and one of a year ago.
Our nightly builds with https://github.com/freifunkh/site/commit/d6ce7f414b117f2b5a1b173b66cc3ccb1a73c741

**Custom patches:**

There are multiple patches in our site repo, whih are not related to the bug ither as stated above:

https://github.com/freifunkh/site/tree/master/patches

AiyionPrime, 2022-03-23:
I do not yet see, why that commit introduced such behavior and will verify the bisect this afternoon.

AiyionPrime, 2022-03-23:
verified: https://github.com/freifunk-gluon/gluon/commit/bbc00017a922fc691297c3ca1263cb7057f63e24 did not contain the error.

AiyionPrime, 2022-03-23:
verified: d728f17b8fc848b35e93eebb8e948798a1e4440a does contain the error and is therefore the first faulty commit.

@MyIgel is informed

MyIgel, 2022-03-26:
Looks like the JS is pretty closely bound to the html structure and browsers add more markup  (`tbody`) when interpretiung it as html5 -> the JS can't find the data or something, have to look further

MyIgel, 2022-03-26:
A first step to a fix would be changing the `Neighbour` function to use sub-childs:
```javascript
var th = iface.table.firstElementChild.firstElementChild;
var el = iface.table.firstElementChild.insertRow();
```

rubo77, 2022-03-27:
Another quick fix would be to revert back to the XML header for just the status-page

AiyionPrime, 2022-03-27:
> The interesting remaining question ist then: What is else broken?

Something I stumbled across was resizing the page to a third of the usual width and see the neighbor section be completely broken.

PR #2457 (2022-04-27): gluon-status-page: Use <tbody> as browsers can add it automatically which broke the page
This fixes #2419 by adding explicit `<tbody>` tags which browsers will add anyhow. An even better option would imho be to use selectors to select elements instead of hard references but that would be a bigger change which should be considered in the future.
closed 2022-04-27 by NeoRaider