freifunk-gluon/gluon #2466: Rename device class "tiny"

lemoer, 2022-04-12:
We just found in the Gluon meeting 2022-02 that there are two different things named identically:
- There is an OpenWrt target called "tiny". (actually ath79-tiny)
- And there is our device class "tiny".

Both things refer to different things, so we should rename our device class to something else.

The new name is open for suggestions.

AiyionPrime, 2022-04-13:
Not sure how useful it is to change it at all; but looking at synonyms for `tiny`, `undersized` might suit the context well.

christf, 2022-04-15:
Are we naming things? 

Possibilities are in cultural references. Small things could be a Gimli or GlodGlodson target. That can be fun but very confusing.

Then we could go explicit: "the small target".

T-Shirt sizes could be an option: XS, S, M, L, XL
Beyond "tiny" we don't use size-based categories though.

Then we could borrow from physics... There are 8 gluons having 8 colours. Red blue green... That would fit the name of the project but it  seems rather unclear what a colour means in this context. Especially when it comes to the anti-color-bits from the model(What is anti red anyway?).

rubo77, 2022-04-19:
low_ram

maurerle, 2024-07-15:
As there actually is no device in gluon build from `ath79-tiny`, and people are using device_class('tiny') in the image-customization.lua - I don't think we should change this.
I would opt for closing this issue instead :)

closed 2024-07-28 by rotanid