freifunk-gluon/gluon #2468: Status Page: multiple data collection processes

herbetom, 2022-04-12:
### Bug report

**What is the problem?**

It can happen that Status Page data collection runs multiple times:

```
gluon-neighbour-info -s  -l -d ::1 -p 1001 -t 3 -r statistics
/lib/gluon/status-page/providers/stations mesh1
/lib/gluon/status-page/providers/neighbours-batadv
```
[Screenshot-htop](https://user-images.githubusercontent.com/15121114/163065163-ca5f91a9-761d-4d5b-9c7c-2605262641b2.png)

It can be reproduced by opening multiple tabs of the status page. But it can also happen with just a single tab and some Browser foo.

The processes seem to accumulate and eventually all memory is used and there is a reboot.

**Gluon Version:**
Seen on Gluon master Branch for quite some time (quite possibly since the bump to 21.03). No reports regarding the last [v2021.1.1](https://github.com/freifunk-gluon/gluon/releases/tag/v2021.1.1) release.

**Reports from the following devices:**
  - ath79:
      - EAP225-Outdoor (128 MB RAM)
      - WS-AP3610 (128 MB RAM)
  - ramips-mt7621:
      - Mi Router 4A Gigabit Edition (128 MB RAM)

AiyionPrime, 2022-05-07:
I think I reproduced the behavior on an or750i on the current master.

AiyionPrime, 2022-06-14:
@herbetom and I met online sunday and dug into this issue.
I do not have a device to test further for now, but tested some commits:

This is the current log in case someone wants to take this further:

```
git bisect start
# bad: [ae7b87b0f0da02e658ee6ea3a910bef3afb2168d] Merge pull request #2520 from freifunk-gluon/check-updater-branch
git bisect bad ae7b87b0f0da02e658ee6ea3a910bef3afb2168d
# bad: [36f406746e88e40ed9b61ae43e99702216970f23] gluon-status-page: fix mesh-vpn section for wg (#2502)
git bisect bad 36f406746e88e40ed9b61ae43e99702216970f23
# good: [a5e085b57c02fe75b3b831719217e8f0efd22bd5] ramips-mt7621: add Xiaomi Mi Router 4A (Gigabit Edition)
git bisect good a5e085b57c02fe75b3b831719217e8f0efd22bd5
# good: [0db3c58329ec17ee09ab2abe1b821c55369512a4] modules: update OpenWrt base
git bisect good 0db3c58329ec17ee09ab2abe1b821c55369512a4
```

On both 'bad' commits I was able to have processes gradually building up, which should not happen.

We did so by having chromium run in background and waited for over an hour or so until it started to happen.

Testing 'good' commits is unfeasible on such a small scale, I think.
Especially since for 'good' commits the bug has just not shown "yet".

Eitherway, just wanted to leave this here, maybe we can go through options later.

rotanid, 2022-07-03:
reopen until packages repo got pushed in the modules file

PR #2572 (2022-07-03): modules: update gluon
04d2b6f sse-multiplex: fix pipe FD leaks

closes #2468 

local build/test successful
closed 2022-07-03 by rotanid