freifunk-gluon/gluon #2470: Configurable LED support

goligo, 2022-04-12:
The latest changes in master lead to regenerating system configuration for every configuration-change or sysupgrade. This also resets any custom LED configuration done by the user.

There are two common use cases for custom LED configuration:
- DIsable all LEDs
- Disable flickering of LEDs (i.e. showing rx/tx of interfaces)

It would be possible for Freifunk communities to create a custom package to achieve this. However it would be useful, if Gluon would provide a configuration property to enable these two use cases, like

gluon.led.mode=default|off|static

As blocktrron pointed out, this should not go into the config UI, as there are LEDs on some devics, which can not be controlled by the firmware, but are directly controlled by the network/wifi chip.

Somebodyisnobody, 2022-05-07:
I am for renaming `default` to `activity` as "default" doesn't is not expressive.

Djfe, 2023-02-11:
I agree, but not all LEDs blink on activity. Activity is the default for some types of LEDs (WiFi), while static is the default for others (power).

maurerle, 2023-08-28:
Some pointers on how to implement this in gluon or as a custom package include the ffgraz-blink package which utilizes `/lib/functions/leds.sh` from openwrt for short term changes of the led behavior:
https://github.com/freifunk-gluon/community-packages/blob/master/ffgraz-blink/files/usr/bin/gluon-blink

If there is demand for a switching behavior throughout the day, with Config-Mode implementation, the ap-timer might be valuable for reference:
https://github.com/FreifunkHochstift/ffho-packages/tree/master/ffho-ap-timer
https://github.com/FreifunkHochstift/ffho-packages/tree/master/ffho-web-ap-timer

I would imagine a custom package which allows to turn off the LEDs from 20 to 8 or something.

maurerle, 2023-10-10:
A possible way which already does this but is also outdated is: ffffm-button-bind - which can set various other things too.
https://github.com/rubo77/ffm-packages/blob/2018.1.x/ffffm-button-bind/README.md

Just for reference

grische, 2026-09-12:
We added a community package that allows turning LEDs off https://github.com/freifunk-gluon/community-packages/tree/main/ffmuc-leds-off