freifunk-gluon/gluon #2471: Interface roles: Removing all roles from an interface causes

goligo, 2022-04-12:
When removing all roles from an interface, to disable it, gluon-reconfigure fails with an error. Typical use case would be to remove all roles from iface_single, and add iface_lan and iface_wan with VLAN interfaces instead.

```
# uci del_list gluon.iface_lan.role='client'
# uci commit gluon
# gluon-reconfigure 
Configuring: 001-reset-uci
cfg030f15
cfg073777
cfg081ec7
cfg01e48a
Configuring: 002-migrate-system
Configuring: 005-set-domain
Configuring: 005-wireless-migration
Configuring: 010-primary-mac
Configuring: 020-interfaces
Configuring: 021-interface-roles
Configuring: 030-system
Configuring: 100-lock-password
Configuring: 110-network
/usr/bin/lua: /usr/lib/lua/gluon/util.lua:25: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function '?'
	/usr/lib/lua/simple-uci.lua:22: in function 'foreach'
	/usr/lib/lua/gluon/util.lua:123: in function 'get_role_interfaces'
	./110-network:17: in main chunk
	[C]: ?
Configuring: 120-ntp-servers
Configuring: 150-poe-passthrough
Configuring: 180-outdoors
Configuring: 200-wireless
Configuring: 210-interface-mesh
/usr/bin/lua: /usr/lib/lua/gluon/util.lua:25: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function '?'
	/usr/lib/lua/simple-uci.lua:22: in function 'foreach'
	/usr/lib/lua/gluon/util.lua:123: in function 'get_role_interfaces'
	./210-interface-mesh:5: in main chunk
	[C]: ?
Configuring: 300-firewall-rules
Configuring: 300-gluon-client-bridge-network
/usr/bin/lua: /usr/lib/lua/gluon/util.lua:25: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function '?'
	/usr/lib/lua/simple-uci.lua:22: in function 'foreach'
	/usr/lib/lua/gluon/util.lua:123: in function 'get_role_interfaces'
	./300-gluon-client-bridge-network:5: in main chunk
	[C]: ?
Configuring: 300-setup-mode
Configuring: 310-gluon-client-bridge-local-node
Configuring: 310-gluon-mesh-batman-adv-mesh
Configuring: 320-gluon-client-bridge-wireless
Configuring: 320-gluon-mesh-batman-adv-client-bridge
Configuring: 320-setup-ifname
Configuring: 325-gluon-private-wifi
Configuring: 330-gluon-mesh-batman-adv-mac-addresses
Configuring: 400-mesh-vpn-wireguard
Configuring: 400-neighbour-info-firewall
Configuring: 400-respondd-firewall
Configuring: 500-autoupdater
Configuring: 500-mesh-vpn
Configuring: 500-node-info-system
Configuring: 500-opkg
Configuring: 500-status-page
Configuring: 505-autoupdater-next
Configuring: 510-autoupdater-wifi-fallback
Configuring: 510-node-info-role
Configuring: 820-dns-config
Configuring: 900-remove-aptimer-config
Configuring: 997-migrate-preserved
Configuring: 998-commit
Configuring: 999-version
One or more upgrade scripts failed. Please review the above error messages.
```

goligo, 2022-04-12:
As a workaround it is currently possible to set a invalid role (e.g. 'none') instead.

NeoRaider, 2022-04-12:
Fix in #2472

PR #2472 (2022-04-16): gluon-core: fix gluon.util.get_role_interfaces() with empty role list
The function failed when an interface has no roles assigned, breaking
several upgrade scripts.

Closes #2471
commit 12528712 (2022-04-16): gluon-core: fix gluon.util.get_role_interfaces() with empty role list (#2472)

The function failed when an interface has no roles assigned, breaking
several upgrade scripts.

Closes #2471
closed 2022-04-16 by blocktrron