freifunk-gluon/gluon #2491: TP-Link WDR4900 does not boot with OpenWrt 22.03

blocktrron, 2022-04-23:
The TP-Link WDR4900 does not boot with a 22.03 based Gluon due to the the kernel exceeding the max bootable size.

@NeoRaider is working on a SPI loader to allow larger kernels to boot. This issue is for tracking said process after the merge of #2426

rotanid, 2022-04-23:
here's neoraiders OpenWrt branch to test a fix:
https://git.openwrt.org/?p=openwrt/staging/neoraider.git;a=shortlog;h=refs/heads/mpc85xx-spi-loader

rotanid, 2022-06-15:
@NeoRaider do you need help testing ?

rotanid, 2022-07-07:
@NeoRaider ~180 devices (src: multi.meshviewer.org) would be happy to get to the next OpenWrt/Gluon release ;-)

rotanid, 2022-09-09:
upstream issue for this problem:
https://github.com/openwrt/openwrt/issues/9027
@pmelange did test a 22.03.0-rebased version of the patches and reported it over there

rotanid, 2022-10-16:
i'd like to reopen this until we have it in a stable branch

AiyionPrime, 2022-10-17:
> i'd like to reopen this until we have it in a stable branch

I do not really mind, but does that imply we only close issues once/if they are backported?

AiyionPrime, 2022-10-17:
> > > i'd like to reopen this until we have it in a stable branch
> > 
> > 
> > I do not really mind, but does that imply we only close issues once/if they are backported?
> 
> It's already backported in OpenWrt, just needs a bump + reenabling the device in Gluon.

That happened in #2675 earlier this afternoon.

I think I might need an introduction to what needs to become a backport, but that's for another time.

pmelange, 2022-10-17:
5db6914f7cfddd1c3cb8a46017d4360e1ec26c5d is the upstream commit in 22.03.
2fa53c9214b7b93fd82ad1ff885145b9e1c1f71b is the upstream commit in master.

rotanid, 2022-10-17:
> > i'd like to reopen this until we have it in a stable branch
> 
> I do not really mind, but does that imply we only close issues once/if they are backported?

if the issue also affects a supported stable branch, yes, i guess, as long as there was no decision to not backport.

PR #2671 (2022-10-15): mpc85xx-p1010: re-enable TP-Link WDR4900
Upstream added a standalone SPI kernel-loader which fixes the unbootable
image for the WDR4900. Thus, we can re-introduce this device to Gluon.

closes #2491
closed 2022-10-17 by rotanid