freifunk-gluon/gluon #2507: Joy-IT OR750i: sysupgrade doesn't work

nrbffs, 2022-05-06:
### Bug report

**What is the problem?**
Upgrading the device doesn't work, neither from autoupdater, nor from config mode or shell via sysupgrade. Serial console is unfortunately also broken in Linux (fixed by 7054721cf94f4aa8fe71ac4c28c922b19a69ba1d ("ath79: enable UART in SoC DTSI files")), so no way to see the full output during upgrade.

**What is the expected behaviour?**
Upgrade works fine.

**Gluon Version:**
2021.1.1
2021.1.2

**Site Configuration:**
https://gitlab.freifunk-stuttgart.de/firmware/site-ffs/-/tree/69443259b8a91ad03785faefe6875e74442cd793

**Custom patches:**
See here: https://gitlab.freifunk-stuttgart.de/firmware/ffs-pipeline-nightly/-/tree/master/patches

AiyionPrime, 2022-05-06:
I think @freifunkh has an increased interest in fixing this.

I asked @ctandi whether he can reproduce the issue with our site;
building a 2021.1.1 based version with a cherry-pick of 7054721cf9 seems favorable, I'll try that this weekend.

@CodeFetch maybe you've got a hunch what went wrong during the update?

nrbffs, 2022-05-07:
I made a build with a working serial console and now the sysupgrade seems to work fine. Is it possible sysupgrade is trying to access the console, fails and thus won't install the upgrade?

AiyionPrime, 2022-05-07:
I do not know about the internals of `sysupgrade`, yet.
But how did you enable the console? Did you manage to patch just that? Or did you update some feed, which fixes countless other things too?

nrbffs, 2022-05-07:
This is the patch (against Gluon) I've used to fix the serial: https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/c79ef2f0703712bd8763edaa54c6cbeac17ed430

I can still get the upgrade to fail sometimes when I start it shortly after the boot:

```
root@(none):/# [   22.368188] urngd: v1.0.2 started.
[   22.747694] random: crng init done
[   29.616927] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   29.624263] eth0: link up (1000Mbps/Full duplex)
[   29.629057] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   29.663719] br-setup: port 1(eth0.1) entered blocking state
[   29.669494] br-setup: port 1(eth0.1) entered disabled state
[   29.675650] device eth0.1 entered promiscuous mode
[   29.680659] device eth0 entered promiscuous mode
[   29.741436] br-setup: port 1(eth0.1) entered blocking state
[   29.747204] br-setup: port 1(eth0.1) entered forwarding state
[   29.753379] IPv6: ADDRCONF(NETDEV_UP): br-setup: link is not ready
[   30.680427] IPv6: ADDRCONF(NETDEV_CHANGE): br-setup: link becomes ready
[   38.207667] jffs2_scan_eraseblock(): End of filesystem marker found at 0x0
[   38.214949] jffs2_build_filesystem(): unlocking the mtd device... 
[   38.214955] done.
Watchdog handover: fd=3
- watchdog -
killall: telnetd: no process killed
Sending TERM to remaining processes ... logd urngd netifd uhttpd dnsmasq sh mount_root sh ubus ubusd 
Sending KILL to remaining processes ... mount_root mount_root mount_root mount_root mount_root mount_root mount_root mount_root mount_root mount_root 
Failed to kill all processes.
s[   38.223366] jffs2_build_filesystem(): erasing all blocks after the end marker... 
ysupgrade aborte[   76.433702] Removing MTD device #5 (rootfs_data) with use count 2
d with return code: 256
[   76.473174] spi spi0.0: error -143 reading 5
[   76.477595] error -143 reading SR
[   76.481080] jffs2: Erase at 0x002d0000 finished, but state != MTD_ERASE_DONE. State is 0x10 instead.
[   76.490528] jffs2: Erase at 0x00280000 failed immediately: errno -143
[   76.497537] reboot: Restarting system
```

Seems like some kind of race between jffs2 and sysupgrade? Not sure what's going on.

Iwwazwersch, 2022-05-07:
Hi, i have the same issue!

nrbffs, 2022-05-07:
My fix is now in #2512

s-2, 2022-05-07:
Same issue here (freifunkh v21, 2021.1.1), console is working in one direction, but won't accept any keystrokes... I had even tried re-soldering that pin, used a different cp2101 adapter and cable etc... :joy:

`sysupgrade -T` did not give any output at all, `-F` did not work either, it would just reboot and stick with the old version.

So I guess waiting for `jffs2` creation to finish will not be sufficient, and this should be flashed via bootloader now.

AiyionPrime, 2022-05-07:
@s-2 have you tested @nrbffs fix?

s-2, 2022-05-07:
I flashed to vH25 using the recovery http interface, and the device is up and running again. As far as I understand, this is a bugfix release still based on OpenWrt 19, so there will be one more manual update required to a 22.03-based version (containing the fix), whenever it becomes available.

Curiously, primary MAC address is incremented by 3, which still leads to a valid address for the device, but will create a duplicate device entry on the map (can be observed with other OR750i as well), also no longer using the label mac address.

nrbffs, 2022-05-08:
Fixed by #2512.

closed 2022-05-08 by nrbffs