freifunk-gluon/gluon #2511: SSH connection only works when enabling legacy key types

RalfJung, 2022-05-07:
### Bug report

**What is the problem?**
On current Debian testing, I cannot just SSH into a gluon node any more. When I try, it says:
```
Unable to negotiate with 2a03:2260:3009:1400:32b5:c2ff:fe6e:9d6b port 22:
no matching host key type found. Their offer: ssh-rsa
```
Looks like ssh-rsa is now considered a legacy key type, and the nodes do not support any sufficiently modern keytype that would still be enabled by default in Debian testing's SSH client.

**What is the expected behaviour?**

Connecting to a gluon node from a Debian testing machine should just work.

Probably this requires updating the SSH server (or at least its configuration) to support modern SSH key types such as rsa-sha2-256.

**Gluon Version:**
v2021.1.1

**Site Configuration:**
https://git.hacksaar.de/FreifunkSaar/gluon-site/-/tree/1.11.0

**Custom patches:**

None

rotanid, 2022-05-07:
this depends on the Dropbear SSH Server Version of the underlying OpenWrt release and therefore is nothing Gluon could change.

Others have this "issue" for a long time when using bleeding edge Linux distributions like Arch Linux and derivatives.

You can use a config like this in your `~/.ssh/config` file to only apply these settings to Freifunk nodes, but not to all connections.
Replace the IPv6 Prefix with the prefix used in your community.
```
Host 2001:470:733f:*
    UserKnownHostsFile /dev/null
    LogLevel QUIET
    KexAlgorithms +diffie-hellman-group14-sha1
    MACs +hmac-sha1
    HostKeyAlgorithms +ssh-rsa
    PubkeyAcceptedKeyTypes +ssh-rsa
Host fd*
    UserKnownHostsFile /dev/null
    LogLevel QUIET
    KexAlgorithms +diffie-hellman-group14-sha1
    MACs +hmac-sha1
    HostKeyAlgorithms +ssh-rsa
    PubkeyAcceptedKeyTypes +ssh-rsa

```

blocktrron, 2022-05-07:
The SSH server used in the upcoming Gluon version / current master supports Ed25519. There is nothing we can do about the current release branches, which use the Dropbear from OpenWrt 19.07.

closed 2022-05-07 by blocktrron

RalfJung, 2022-05-07:
Is there a way to document this more prominently? It took me a bit to figure out that this is not a network issue or a node issue, but that I have to adjust my local SSH confug.