freifunk-gluon/gluon #2517: Xiaomi AX3200 support

burnziii, 2022-05-11:
Hi,

is the Xiaomi AX3200 already supported within gluon?
OpenWRT seems to be working fine (https://openwrt.org/toh/xiaomi/ax3200) and the price for this device is very good.
I haven't found anything at readthedocs.

Regards,

Christoph

belzebub40k, 2022-05-11:
This does not sound very promising

> At the present moment (April 2022) there is no known way to enable telnet on the Xiaomi AX3200 (model RB01, international version). So if you have this device, you can check if it came from factory with telnet enabled. If so you can install OpenWrt, otherwise it is not possible to install OpenWrt on the Xiaomi AX3200 (model RB01). 

https://openwrt.org/toh/xiaomi/ax3200#installation

blocktrron, 2022-07-07:
Closed in #2566

closed 2022-07-07 by blocktrron