freifunk-gluon/gluon #2518: 2021.1.2: libubus20191227 dependency for dnsmask not satisfied

smoe, 2022-05-14:
Hello,
after "make update" from 2021.1.1 and our routine build skript I reproducibly run into 
```
Configuring luabitop.
Configuring luaposix.
Configuring gluon-core.
Configuring urngd.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for dnsmasq:
 *      libubus20191227
 * opkg_install_cmd: Cannot install package dnsmasq.
make[3]: *** [package/Makefile:68: package/install] Error 255
make[3]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make[2]: *** [package/Makefile:108: /home/moeller/gluon/gluon_2021.1.x/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2
make[2]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make[1]: *** [/home/moeller/gluon/gluon_2021.1.x/openwrt/include/toplevel.mk:227: world] Error 2
make[1]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make: *** [Makefile:185: all] Error 2
```
I then also tried the latest master for which I did not see that error but
```
Setting flags for package libc to hold.
Setting flags for package libgcc1 to hold.
Setting flags for package libpthread to hold.
cp -fpR /home/moeller/gluon/gluon_2021.1.x/openwrt/build_dir/target-mips_24kc_musl/root-ath79 /home/moeller/gluon/gluon_2021.1.x/openwrt/build_dir/target-mips_24kc_mus
l/root.orig-ath79
/home/moeller/gluon/gluon_2021.1.x/openwrt/build_dir/target-mips_24kc_musl/root-ath79/usr/lib/opkg/info/gluon-core.postinst-pkg: line 88: 2917062 Segmentation fault   
   lua "$IPKG_INSTROOT/lib/gluon/check-site.lua" <<'END__GLUON__CHECK__SITE'
need_string(in_site({'site_code'})) 
need_string(in_site({'site_name'})) 
...
END__GLUON__CHECK__SITE

postinst script ./usr/lib/opkg/info/gluon-core.postinst has failed with exit code 139
make[3]: *** [package/Makefile:73: package/install] Error 1
make[3]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make[2]: *** [package/Makefile:111: /home/moeller/gluon/gluon_2021.1.x/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_install] Error 2
make[2]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make[1]: *** [/home/moeller/gluon/gluon_2021.1.x/openwrt/include/toplevel.mk:230: world] Error 2
make[1]: Leaving directory '/home/moeller/gluon/gluon_2021.1.x/openwrt'
make: *** [Makefile:194: all] Error 2
```
It all built fine when I newly checked out master from github in a new directory and then checked out the v2021.1.2 tag.

I still have the old directory. If anyone has any clue for me where to start looking - tell me.

A-Kasper, 2022-05-14:
I had the same problem. After make dirclean it worked. Maybe it's the same in your case.

NeoRaider, 2022-05-15:
I was able to reproduce the problem on my machine. I have not checked if Gluon master is still affected (or might be affected with future package updates), but I intend to look into it.

Note that a clean Gluon tree is necessary when switching OpenWrt versions (for example Gluon v2021.1.x to master), but an update to a newer version of v2021.1.x without switching the branch should not cause such issues, so there is definitely a bug.