freifunk-gluon/gluon #2519: Autoupdate Not working if branch is not in site

rubo77, 2022-05-14:
We accidentally offered a firmware in Freifunk Nord that had the Autoupdate branch defined to "2020.1.x" instead of "stable".

Since there's no firmware mirror defined for that, the au is not working.

It would have been helpful, if the au would fall back to stable if the branch is not defined

rubo77, 2022-05-15:
To the user this problem is not visible in config Mode, since the select Box only shows "stable" because that is the first option.

There should be a warming there, that in uci there is set am invalid branch

rubo77, 2022-05-15:
Will that commit also show a sensible display on misconfigured nodes in the admin panel?
E.g. show the configured branch somewhere instead of quietly just display the first entry in the selection?

NeoRaider, 2022-05-15:
My PR will only prevent such a firmware from being built so far.

I don't think the UI needs a warning - instead we should make sure that node can't even get into the state where an invalid branch is set (unless someone manually edits the UCI config). One option would be to revert to the default branch on updates when an invalid branch is configured as an additional safeguard against misconfiguration, for the rare cases where
- the list of branches has actually changed or
- a firmware with broken default branch was manually updated.

rubo77, 2022-05-15:
yes, that and the autoupdater should also check, if the configured branch is in the site.conf and if not, use the default branch to check for download.

ATM the autoupdater does not work at all if the branch is not in the list

PR #2520 (2022-06-11): gluon-autoupdater: check default branch name, check number of pubkeys, revert invalid branch on update/reconfigure
Closes #2519
Closes #2532

Checked build with:
- Valid branch: OK
- Unset branch: OK
- Invalid branch: Build failed as expected
closed 2022-06-11 by NeoRaider