freifunk-gluon/gluon #2531: Requesting adoption of Netgear Nighthawk X6 r8000

smoe, 2022-05-24:
Hello,

OpenWrt describes the r8000 as working (https://openwrt.org/toh/netgear/r8000) , even though the Broadcom chipset would not upfront allow to expect that. I wanted to give it a try. However, I utterly failed to integrate the building of that bcm53xx target (already found in 19.07) with gluon.

Could you please add this machine as "broken, untested" to Gluon or give me a pointer how to integrate a new target? I had added new devices for existing targets before and somehow it is more than just creating that file and adding the include in targets.mk.

Many thanks!

herbetom, 2022-06-14:
You did already notice the few notes withing https://gluon.readthedocs.io/en/latest/dev/hardware.html#adding-support-for-new-hardware-targets regarding adding new targets?

blocktrron, 2022-07-03:
@smoe Please provide a link to your branch where you've tried to add the target.

FWIW, Broadcom is a dead end considering wireless support, so I will close this issue for this reason.

closed 2022-07-03 by blocktrron

smoe, 2022-09-11:
@blocktrron,

> @smoe Please provide a link to your branch where you've tried to add the target.

Thank you for what I interpret as an implicit offer of yours to have a look. I had changed/tried so much that I could not provide anything I would want you to investigate, sorry for that extra time to get back to you. In the background, I now have 2022.1 compiling and as soon as time permits will revisit my shot at the r8000 and then drop the URL here.
 
> FWIW, Broadcom is a dead end considering wireless support, so I will close this issue for this reason.

Please check out the comment on https://openwrt.org/toh/netgear/r8000 that states that the [brcmfmac](https://openwrt.org/docs/techref/driver.wlan/brcmfmac) driver in the kernel is well supported and hence this device an exception to the rule. If my understanding is not completely off then p´lease kindly consider to reopen this issue.