freifunk-gluon/gluon #2571: cc: Permission denied

Byggvir, 2022-06-29:
When building a new image I get the error 

make[2]: Entering directory '/home/thomas/gluon/multi-domain/v2021.1.2/openwrt/scripts/config'
**make[2]: cc: Permission denied**

```
Updating goal targets....
Considering target file 'defconfig'.
 File 'defconfig' does not exist.
  Considering target file 'scripts/config/conf'.
   File 'scripts/config/conf' does not exist.
    Considering target file 'staging_dir/host/.prereq-build'.
      Considering target file 'include/prereq-build.mk'.
       Looking for an implicit rule for 'include/prereq-build.mk'.
       Trying pattern rule with stem 'prereq-build.mk'.
       Found an implicit rule for 'include/prereq-build.mk'.
       Finished prerequisites of target file 'include/prereq-build.mk'.
      No need to remake target 'include/prereq-build.mk'.
     Finished prerequisites of target file 'staging_dir/host/.prereq-build'.
     Prerequisite 'include/prereq-build.mk' is older than target 'staging_dir/host/.prereq-build'.
    No need to remake target 'staging_dir/host/.prereq-build'.
   Finished prerequisites of target file 'scripts/config/conf'.
  Must remake target 'scripts/config/conf'.
Putting child 0x564bb8c5a4f0 (scripts/config/conf) PID 118198 on the chain.
Live child 0x564bb8c5a4f0 (scripts/config/conf) PID 118198 
make[2]: Entering directory '/home/thomas/gluon/multi-domain/v2021.1.2/openwrt/scripts/config'
make[2]: cc: Permission denied
make[2]: *** [<builtin>: conf] Error 127
make[2]: Leaving directory '/home/thomas/gluon/multi-domain/v2021.1.2/openwrt/scripts/config'
Reaping losing child 0x564bb8c5a4f0 PID 118198 
make[1]: *** [/home/thomas/gluon/multi-domain/v2021.1.2/openwrt/include/toplevel.mk:116: scripts/config/conf] Error 2
Removing child 0x564bb8c5a4f0 PID 118198 from chain.
make[1]: Leaving directory '/home/thomas/gluon/multi-domain/v2021.1.2/openwrt'
Sammle erfolglosen Kindprozess 0x55b0bebe5030 PID 118072 
make: *** [Makefile:157: openwrt/staging_dir/hostpkg/bin/lua] Fehler 2
Entferne Kindprozess 0x55b0bebe5030 PID 118072 aus der Kette.

```

AiyionPrime, 2022-06-29:
hey @Byggvir, thanks for reporting this.
There's an issue-template for Bug reports iirc, you could use that next time.

Since you did not this time, you are missing several important details in your problem description. Please take a look at this and update your issue accordingly.

https://github.com/freifunk-gluon/gluon/issues/new?assignees=&labels=&template=bug_report.md

Thanks!

NeoRaider, 2022-06-29:
I don't think anything in the template would be particularly helpful in this case... I'm more interested in the outputs of the following commands:
- `command -v cc`
- `cc --version`

AiyionPrime, 2022-06-30:
Alright, taking a guess as well.
His installed compiler might just be fine, regardless of its version.

@Byggvir OpenWrt evaluates `$PATH` using `which` in order to determine the path of the compiler to use. If your `$PATH` would reference a directory which contained a file/directory called `gcc`, the Makefile would try to execute that instead...

NeoRaider, 2022-06-30:
> Alright, taking a guess as well. His installed compiler might just be fine, regardless of its version.

Yes, that's my assumption as well. `cc --version` is just an arbitrary command that should always success if `cc` is a working compiler at all.

AiyionPrime, 2022-07-03:
@Byggvir Could you post the output of the commands you tested?

AiyionPrime, 2022-07-10:
Alright @Byggvir, time to get this sorted out.
Can you open a terminal and put the output of them here?

AiyionPrime, 2022-07-17:
I think we can close this. I failed to reproduce the issue with gluons current master;
and @Byggvir did not answer for another week now...

closed 2022-07-17 by NeoRaider