freifunk-gluon/gluon #2592: add an internet connectivity scan to the status page

rubo77, 2022-08-02:
On the status page there is a graph for the connectivity history to all neighbours.

It would be great if we had an extra line that pings an URL in the internet

AiyionPrime, 2022-08-03:
Just leaving this here.
https://github.com/freifunk-gluon/gluon/issues/2228

rubo77, 2022-08-03:
@AiyionPrime you Treffer to this comment?

> having all nodes ping into the world in regular intervals is not something we would like to see

My idea here is something different: i want to ping only from the status page witch is usually not open anywhere unless you manually open it in your browser, so there will be no mentionworthy traffic generated by this new feature of the status page

AiyionPrime, 2022-08-04:
> i want to ping only from the status page

I did not expect that, yes.
Dont't you think it might be misleading to have a page full of device-related info as well as a ping check, that shows whether your client can ping heise, or whatnot?

Assuming I am at home, connected via telekom and access the statuspage of the router in our hackerspace. That ping would show me how reliable icmp works from telekom to heise at my place?
Or am I missing something?

rubo77, 2022-08-04:
Your right, That would be nearly useless,

What I mean is, that the node should ping the internet, while the status page is viewed

AiyionPrime, 2022-08-06:
I still have doubts whether that would be a good idea.

What you are proposing still means, I can grab curl, hit up our 2000 nodes' statuspages and they will all ping heise or github from over thousand different endpoints.
That's a proper proxy mechanism.

rubo77, 2022-08-08:
We could easily work around this DOS exploit if we enable the ping scan only during the first few hours after a reboot

AiyionPrime, 2025-08-26:
@rubo77 We discussed pinging to be too costly at length in the other issue linked above, which led to the development of `gluon-state-check`.

While this would allow for a package to provide ping-checks, I doubt this would ever get past community packages due to the imbued proxy mechanism :/

If another owner or you're still interested in developing this, let me know.
I'll gladly reopen this issue.

Please hit me up, if you'd like technical assistance.

closed 2025-08-26 by AiyionPrime