freifunk-gluon/gluon #2619: Missing kmod-ath10k on Netgear R7800

mweinelt, 2022-08-22:
Problem introduced in https://github.com/freifunk-gluon/gluon/commit/15ef885836907b27b2c0481f7ea83d1dd504c5d5.

After install kmod-ath10k and modprobing `ath10k-core` and `ath10k-pci` the phys appear. Rebooting the device post-upgrade makes it then work again normally.

NeoRaider, 2022-08-23:
The image should contain `kmod-ath10k-ct`, not `kmod-ath10k`.

At least after `make GLUON_TARGET=ipq806x-generic config`, `openwrt/.config` contains `CONFIG_PACKAGE_kmod-ath10k-ct=y`. A full test build of *ipq806x-generic* is currently running on my machine.

NeoRaider, 2022-08-23:
My build finished, and it contains `ath10k_core.ko` and `ath10k_pci.ko` from kmod-ath10k-ct.

@mweinelt Either you have a site-specific issue, or the device is not working correctly with the CT driver. Can you check which is the case?

mweinelt, 2022-08-23:
The kmod is missing from our testing images since 2022-06-05. I'm not sure what causes it.

https://git.darmstadt.ccc.de/ffda/firmware/site

PR #2621 (2022-08-25): Include OpenWrt's target default package lists in config generation
See the commit message of patch 3 ("scripts: target_config_lib: prepend target default package list from openwrt/tmp/.targetinfo") for an explanation of the issue.

Patch 1 ("scripts: target_config_lib: remove handling for targets without devices") looks much more complex that it is due to unfortunate diffing - it only removes the obsolete `if #lib.devices > 0` check and the corresponding else branch.

I have diffed the generated `.config` for all targets with BROKEN=1 before and after the change (filtering out `CONFIG_GLUON_V [... cut]
closed 2022-08-25 by blocktrron