freifunk-gluon/gluon #2630: mpc85xx-p1010: waiting for user input during build process

belzebub40k, 2022-09-07:
### Bug report

**What is the problem?**
The build process for the mpc85xx-p1010 target stops and waits for a user input to select the CPU.

```
  ...
  HOSTLD  scripts/kconfig/conf
.config:1108:warning: override: CPU_LITTLE_ENDIAN changes choice state
*
* Restart config...
*
*
* Processor support
*
Processor Type
  1. 512x/52xx/6xx/7xx/74xx/82xx/83xx/86xx (PPC_BOOK3S_6xx)
> 2. Freescale 85xx (PPC_85xx)
  3. Freescale 8xx (PPC_8xx)
  4. AMCC 40x (40x)
  5. AMCC 44x, 46x or 47x (44x)
  6. Freescale e200 (E200)
choice[1-6?]: 2
CPU selection
> 1. e500 (8540) (E500_CPU) (NEW)
  2. Rely on the toolchain's implicit default CPU (TOOLCHAIN_DEFAULT_CPU) (NEW)
```

**What is the expected behaviour?**
The build process should run without any user interaction

**Gluon Version:**
Current master (v2022.1-5-g19b54dd5)

**Site Configuration:**
https://github.com/freifunk-mwu/site-ffmwu (e09e533)

**Custom patches:**
None

herbetom, 2022-09-15:
I've just encountered a stuck build of the `mpc85xx-p1020` target. The last line was `HOSTLD  scripts/kconfig/conf` but i didn't get a menu for whatever reason.

Just building it again resulted in the same behaviour. Then i tried `make V=s BUILD_LOG=1 -j1` and it ran through without a problem. And after that the parallel build also succeeded again.

blocktrron, 2022-10-10:
Should be resolved with current OpenWrt base.

closed 2022-10-10 by blocktrron

belzebub40k, 2022-10-11:
Thanks. I can confirm it is working again for mpc85xx-p1010 and mpc85xx-p1020.

PR #2658 (2022-10-07): [v2022.1.x] modules: update to latest HEAD
This bumps the modules to the same point as #2657 does it for master.

I've created this PR because i would like to have the Commit which hopefully fixes #2630 in the v2022.1.x release branch.