freifunk-gluon/gluon #2655: Boot issues with 2022.1 on TP-Link-WDR3600 and TP-Link-WDR4300

Grotax, 2022-10-04:
### Bug report

**What is the problem?**
Booting on TP-Link-WDR3600 and TP-Link-WDR4300 is failing often (about 50/50) since the update to 2022.1 coming from v2021.1.x.
With a normal power off and on cycle the power LED will turn on, nothing else happens after that.
The issue was noticed for the first time during a normal reboot via the command line, the device did not come up again.
It seems not to affect any other devices.
Clean flash did not help, updating to the same version without keeping the configuration caused the same issue.

**What is the expected behaviour?**
Boot normally after reboot or power cycle

**Gluon Version:**
v2022.1

**Site Configuration:**
https://github.com/ffsh/site/commit/23741772fa48eb9ceb4a2288e6789da8228a8753
[2022.1.0.0](https://github.com/ffsh/site/releases/tag/2022.1.0.0)

Site.mk is long but as almost everything is commented out these routers do not receive any extra packages apart from the standard stuff.

**Custom patches:**
none

The issue was reported by one of our users. Unfortunately I do not own such a device otherwise I would have tried to reproduce it.
We have a couple of devices that are affected by this in a mesh network.
For example [this](https://map.freifunk-suedholstein.de/#!/de/map/e894f6bbc3e2) and the surrounding nodes.
Workaround at the moment is to simply not touch them because they do in fact run fine as long as they are note restarted.

rubo77, 2022-10-06:
Any other communities report this issue?
Do the routers work fine again after downgrading?

Grotax, 2022-10-06:
Downgrading was not tested.
I don't know if other communities have this too. After a quick check it seems like many communities are still on older releases.
So there are not many devices on that version.

Grotax, 2023-01-19:
Since it did not happen again and there was no further investigation possible this issue can be closed.

closed 2023-01-19 by Grotax